volumio/Volumio2 #1545: Stored wifi passwords should not be visible

ioparaskev, 2018-05-05:
Using volumio2 for the past weeks I noticed a scenario where one might be able to infiltrate into your wifi without you noticing. 

Consider the following scenario: Volumio is connected to your wifi and hotspot fallback option is also enabled (default). This can provide a window of opportunity to an attacker in two ways:
1. Setup a fake AP with the same SSID as your AP. The attacker might be able to fool volumio to try and connect to their AP resulting in not being able to connect (since the password will not be correct)
2. Start a storm of deauth attacks to prevent volumio from connecting to the wifi AP

In both cases, since volumio has hotspot fallback option enabled, a hotspot named Volumio with show up. Now if you've stayed on the default options, this will mean that the hotspot password is volumio. So the attacker can connect to it with some googling. No big deal until now besides the fact that your volumio box has gone rogue.
There is a much bigger problem though. When the attacker connects to your volumio hotspot, they can go to volumio network settings and see the password for every wifi volumio has connected to. This means that the attacker now has a clear password view of your wifi AP.

I don't think this scenario is too extreme since it is based on default settings. My suggestion is to remove the option to clear view the wifi password when the password has been submitted (clicked save).

NiekertDev, 2018-05-16:
As an attacker, you Google a bit and enable SSH, SSH into the system and run ```sudo cat /etc/wpa_supplicant/wpa_supplicant.conf```. This way your suggested idea doesn't help. 

Short after, I realized that their can be an solution. Before Volumio saves the password, run it through ```wpa_passphrase```. As from what I know, it hashes the password. [Not Tested]

volumio, 2018-05-16:
Both good solutions. We're currently really struggling with various system upgrades, it would be cool if a PR is submitted for those

ioparaskev, 2018-05-17:
@NiekertDev running it through wpa_passphrase will only hash the password. Someone who will have access to volumio through ssh will simply open the `wpa_supplicant.conf` and copy the hashed psk. Then they can configure their pc to connect to the wifi (by copying the specific entry to their conf file. Running the password through the wpa_passphrase means that you won't have a clear text view of the wifi password, it doesn't mean that someone will not be able to connect.

If you want to verify it, run the following command to 2 different machines:
`wpa_passphrase "test" "lalaalalalallalal"`
the resulting configuration will always be:
```
network={
        ssid="test"
        #psk="lalaalalalallalal"
        psk=d102b2a76e1791efd17d57a0a60946dfb85c5806634ee6d50e25bb5d9a423291
}
```
So someone who is smart enough to enable ssh access to volumio and find the `wpa_supplicant.conf` will always be able to connect to your AP

NiekertDev, 2018-05-17:
Okay, got it

WarmChocolateCake, 2019-08-04:
Just to add that 2.599 still has this issue