volumio/Volumio2 #1546: It is virtually impossible to connect to WiFi with a receive-only dongle

cormacrelf, 2018-05-07:
I have a Pi Zero (not W) with a WiFi dongle. It does not seem to be capable of creating a hotspot, so just sits silently on boot. I also have an apparently transmit-only device which creates a hotspot but doesn’t permit configuring an SSID/PSK such that it would no longer be necessary; the UI just doesn’t have the config sections referred to in the documentation. Further, they don’t work in combination.

I have of course tried configuring `/etc/network/interfaces`, but this creates issues with the UI making AJAX requests to a hotspot IP address (192.168.211...) instead of just using the host name (volumio.local) from the JS Window object (which would solve this entire thing). This is despite there being no Volumio WiFi network visible.

When configuring through the JSON files in /data/ to mirror what the code appears to use and what people on the forum post, I can only get Volumio to connect using those credentials when I force it to create a hotspot. Unfortunately, because the device is receive-only, Volumio connects but just seconds later decides that the hotspot doesn’t work so it will just turn off the entire wlan0 interface. I have watched all this through `journalctl` countless times. You can subsequently re-enable it via the shell (`ip link set dev wlan0 up`), but this is not appropriate for a set-and-forget media centre device. And the various components still try to advertise themselves on the hotspot IP, so no AirPlay/discovery services work anyway.

Am I missing something? Is it possible to bypass the entire hotspot section of the code, so it won’t mess up routes, change the dumb manual address for UI AJAX, or disable wlan0 entirely?

xipmix, 2018-05-07:
Can you see a way to probe the card and determine if it is receive-only? If so wireless.js (in the Build project) could be modified to test that before attempting to start hotspot.

macmpi, 2018-05-07:
Yes PiZero initial setup can be difficult due to lack of built-in networking port.
I assume you do not have a usb hub into which you could simultaneously plug a usb-ethernet dongle to do initial setup (and disable hotspot and set SSID/pass)?

Can you elaborate on which specific dongle you use (chipset & driver), and which Volumio version you use?

cormacrelf, 2018-05-09:
##### Receive-only

* https://thepihut.com/products/usb-wifi-adapter-for-the-raspberry-pi
* model: Ralink Technology, Corp. RT5370 Wireless Adapter
* driver: rt2800usb (https://wiki.debian.org/rt2800usb)

##### Transmit-only

This device looks very different (black, unbranded, with a 8cm antenna) but it is apparently the exact same device, the Ralink 5370.