volumio/Volumio2 #1547: Yet another NAS mount problem

bebek, 2018-05-11:
Hello all, 

I am new to volumio. I have been using it with success and I am happy so far.
My version is the 2.389
I have been listening to music with the INTERNAL storage capability, the USB way as well. The
web radios work nicely.

Now I would like to get access to a network disk. I have been struggling for a while without any
success. I can access this disk without any problem from various windows computers.

Reading volumio forums I saw people did have similar problems solved with "vers=1.0 or 2.0"  flag.

Unfortunalely this does not work for me.

I tried to look at the logs and I found this:
 CIFS VFS: Error connecting to socket. Aborting operation.
 CIFS VFS: cifs_mount failed w/return code = -115

Is there anyone who can give me a clue on how to solve this CIFS error ?

thanks a lot,
Bebek

xipmix, 2018-05-12:
Can you post the log https://volumio.github.io/docs/User_Manual/Sending_logs_for_troubleshooting.html ? The thing we need is how 'mount' is called before it fails.

bebek, 2018-05-13:
Some news ...  I cleared the data with the UI command. Volumio restarted and I redid the configuration and guess what ...
I managed to access to my network drive. However I did not managed to access a windows share on a PC.
Today after reboot Volumio did not manage to connect to the network drive. Same error again the 115.
I posted the logs, here is the link http://logs.volumio.org/volumio/kwqHmm2.html

Let me know if you need more information to figure out where the problem is.

xipmix, 2018-05-14:
Thanks for the log. You see this command:
```
/bin/mount -t cifs -o guest,ro,dir_mode=0777,file_mode=0666,iocharset=utf8,noauto,soft,vers=1.0 //SHASERVEUR/Musique-288AB9CB8AB /mnt/NAS/Musique
```

Can you connect as 'guest' to the share from other machines?
If so, can you try running the command from the command line on volumio, with the -v flag as well:
```
volumio@volumio$ mkdir /tmp/test
volumio@volumio$ sudo /bin/mount -v -t cifs -o guest,ro,dir_mode=0777,file_mode=0666,iocharset=utf8,noauto,soft,vers=1.0 //SHASERVEUR/Musique-288AB9CB8AB /tmp/test
```
If it fails, can you paste the output here? It might be worth trying a few variations, eg ```vers=2.0``` instead of ```vers=1.0```

Next question: does the name 'SHASERVEUR' resolve properly? I would expect to see something like this:
```
volumio@volumio$ host SHASERVEUR
shaserveur.local has address 192.168.1.10

or

volumio@volumio$ nmblookup SHASERVEUR
192.168.1.10 SHASERVEUR<00>
```

bebek, 2018-05-14:
Very strange 

I deleted user data again. Went thru the setup.

Volume webradios do not appear ...  ??

And the network drive is connected with vers=1.0 flag

host does not give an answer 
     volumio@volumio:~$ host SHASERVEUR
     ;; connection timed out; no servers could be reached
however nmblookup returns
     volumio@volumio:~$ nmblookup SHASERVEUR
     192.168.0.5 SHASERVEUR<00>

closed 2018-05-14 by bebek