volumio/Volumio2 #1548: undefined language_code causing app to crash

SylannBin, 2018-05-12:
Original context:

The Volumio webapp stopped working several seconds after I reached the page for the first time on this install.

While connected to the rpi via ssh, I tried:

    volumio status  // 

then I tried

    volumio vrestart //

After spending a lot of time trying to find problems in my rpi, I decided to check volumio itself.

I have then found this recurring error in the log:

```
Error: /volumio/app/i18n/strings_undefined.json: ENOENT: no such file or directory, open '/volumio/app/i18n/strings_undefined.json'
    at Error (native)
    at Object.fs.openSync (fs.js:641:18)
    at Object.fs.readFileSync (fs.js:509:33)
    at Object.readFileSync (/volumio/node_modules/jsonfile/index.js:67:22)
    at CoreCommandRouter.loadI18nStrings (/volumio/app/index.js:1566:25)
    at new CoreCommandRouter (/volumio/app/index.js:65:10)
    at Object.<anonymous> (/volumio/index.js:42:21)
    at Module._compile (module.js:570:32)
    at Object.Module._extensions..js (module.js:579:10)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
    at Module.runMain (module.js:604:10)
    at run (bootstrap_node.js:389:7)
    at startup (bootstrap_node.js:149:9)
    at bootstrap_node.js:504:3
```

Starting the app myself with `/volumio/index.js` gave this exact same error.

Looking at the code I eventually reach the part CoreCommandRouter.loadI18nStrings part...

And here is what I see:

```
    var self=this;
    var language_code=this.sharedVars.get('language_code');

    this.logger.info("Loading i18n strings for locale "+language_code);

    this.i18nStrings=fs.readJsonSync(__dirname+'/i18n/strings_'+language_code+".json");
    this.i18nStringsDefaults=fs.readJsonSync(__dirname+'/i18n/strings_en.json');

    // ...
```
Since `language_code` is not defined in the `sharedVars` (for some reason), the local variable of the same name is also undefined. And this part of the program tries to read a file named `/i18n/strings_undefined.json`

Reading a file synchronously and not handling an error that begs to occur kind of strikes me.

xipmix, 2018-05-14:
Great analysis. What language did you configure please?

volumio, 2018-05-14:
This is the first time I see this issue, but we changed some stuff lately that could have impacted what you see here. Of course, the try catch is a way not to make the whole thing explode, but I'm rather interested to find the root cause. 
Was it a fresh install? If not, what did you do before? Which device do you have?

SylannBin, 2018-05-15:
Thanks fir you replies,
@xipmix I added fr_FR Utf8 only. But I did it especially because I sometime use an azerty keyboard and had problems with some LC variables. (I had the same problems on another raspberry with raspbian installed on it.)

--
@volumio it was a fresh install.
But I also installed various packages like vim, raspi-config, other utils AND .... I installed a gitlab server.

xipmix, 2018-06-22:
I tried to reproduce this with 2.413.
I had a browser tab open on volumio but the pi was turned off so just the waiting spinner showed.
I inserted a freshly flashed card and booted. The browser tab refreshed.
I went to 'appearance' and selected language 'Francais' and saved.
The system them displayed the first run wizard, which asked (in French) which language I wanted (English was selected). I selected Francais again.
So far, so good, no crashing (the vulnerable code is still there).

I checked the available locales on the system
```
$ locale -a
C
C.UTF-8
en_US.UTF-8
```
I turned on fr_FR.UTF-8
```
$ sudo dpkg-reconfigure locales
```
I left the default locale set to 'none', and rebooted.
Boots ok, no crash. Try selecting a song & playing, all good.

Try changing the default locale to fr_FR (again with ```dpkg-reconfigure locales```), reboot.
Now, when I check the volumio processes I see that the LANG env var is set. It was not before.
```
$ cat /proc/<pid>/environ
LANG=fr_FR.UTF-8PATH=/usr/local/sbin:/usr/local/bin:...etc
```
If I try to set another language, eg Deutsch, it works ok and I can go back to Francais with no issue.

So with a 'vanilla' volumio environment this is difficult to reproduce. I suspect something is up with the environment your volumio is starting in, that is messing up the ```sharedVars``` value for ```language_code```.
Unfortunately I cannot see how the sharedVars get set the first time, before there is  a /data/configuration/miscellanea/appearance/config.json file.

xipmix, 2018-06-22:
If you can get these crashes again it would be interesting to know what ```cat /proc/<pid>/environ``` of one of any still-running volumio processes shows.

xipmix, 2018-06-22:
I wrote a patch similar to the example code and tested by renaming the strings_fr.json file and restarting
```
$ sudo systemctl restart volumio
```
I found that some of the the web UI is still displayed in French - the browse/play/queue buttons along the bottom. This persisted even after a shift-reload of the browser tab. The other pages went back to English when loaded. The system locale was still defaulted to fr_FR.UTF-8 but I don't think that matters. If I get time I'll file a separate bug for this.

xipmix, 2018-06-23:
Strange, the UI was all in French again. I went to Appearance and reselected English, and reloaded. Now the whole UI is correctly in English.