volumio/Volumio2 #1550: Multiroom Not Working in v2.389 (RPi)

RMichaelPickering, 2018-05-23:
It appears that devices running image version 2.389 don't have multiroom working properly. Two devices running this image won't detect each other at all - the web UI multiroom section remains empty. However, when a device running an older version, e.g. 2.368, comes online, both it and a device running v.2.389 WILL detect each other and both will appear in the multiroom section.

volumio, 2018-05-23:
Hi Michael,
are the build downloaded from Volumio or self built?
Does the problem persist if you restart them, after having restarted the router?
Wireless or ethernet?

RMichaelPickering, 2018-05-26:
We figured out the issue. Thanks!

volumio, 2018-05-26:
Since this is an issue tracker and not a support ticketing, I would appreciate if you could share some info in order to fix this bug

RMichaelPickering, 2018-05-26:
OK, here you go!

First, a bit of background: I'm building a custom version of Volumio, and for which I'm not using the standard Volumio build process. This includes making multiple copies of the system image after it's been built.

The issue that I encountered was that the multiroom section of the web UI wasn't appearing even though I had several systems running my build.

The reason for this proved to be that the UUID of each system was the same, so even though they each had a different system name and IP address assigned, the multiroom code didn't get triggered because it is using the UUID as the unique identifier of each system.

In case anyone else is hacking Volumio and chooses not to follow the standard, Volumio-recommended build process, it will be important to implement some alternative method for assigning a UUID as per the OSF standard, also known as IETF RFC 4122. Various existing implementations of this standard are readily available, including a node.js library (npm uuid) and a Linux command line utility (uuidgen) which is available in most modern distros.

If my memory serves me correctly, the Volumio configuration file that contains the UUID for the system is located in /data/configuration, and is named something like config.json.

I will leave this issue open to give a chance for further review of this comment, but as I mentioned, this is no longer an open issue and was not actually a problem with Volumio itself -- it turned out to be entirely an issue with my non-standard build process.

volumio, 2018-05-27:
Thank you for the insights, yes this can be useful to others. This is an issue we generally see when an instance of volumio is cloned and used into other systems.
Couple of suggestions: the settings to which you refer to are meant to be generated dynamically: the firsts time, a new UUID is assigned using node-uuid, and this happens also for other parts of the system so I suggest not to copy\clone the settings.

RMichaelPickering, 2018-05-27:
I certainly understand your position on NOT trying to simply clone an image! There is indeed much setup that is done automatically when the autoconfig is done as part of installing the standard image.

Given that you've encouraged me to add information to this issue in case it is useful to others, it would probably be worth adding here that the code you referenced that would automatically assign a uuid would, I believe, be triggered on Volumio startup if the UUID section, including the assigned value, was simply removed from the configuration file.

I would think that cloning an image would be most helpful in case a user wants to set up a multiroom system based on several Volumio instances, and in case the hardware is identical on each system, I believe that this process could be made to work fairly easily by making the modification that I suggested above, after running through the installation process on the first system.

Perhaps we should create a 'how to' that would document this?

volumio, 2018-05-27:
@RMichaelPickering it does work this way: if the config file is removed, or the value of uuid is removed, a new one will be generated. 
There are some other unique identifiers used in the system which are based on hardware fingerprinting, to ensure uniqueness.

The risk I foresee when cloning is that you're getting many other unique identifiers duplicated on the OS level: dhcp UID, SSH keys and so on.

Bottom line: I suggest using a build script to generate images, instead of cloning instances to modify the system to fit some purposes

RMichaelPickering, 2018-10-24:
This wasn't an actual issue with Volumio so we should close it.

closed 2018-10-24 by RMichaelPickering