volumio/Volumio2 #1551: Resampling of Shairport-Sync output

MKSounds, 2018-05-24:
Hello,

I'm using the I²S output of a Raspberry Pi 2b with Volumio 2.389 to provide audio data to a dsp (ADAU1701). The dsp must be configured as clock-master (so the raspberry is a clock slave) and because it doesn't have an ASRC, the input sample rate has to be fixed at 48 kHz (because of the mounted oscillator).
I use my overlay in combination with the onboard resampling of volumio. In this case everything works perfect with local media.

Now I want to use AirPlay to stream my audio to volumio, but the output of shairport-sync is limited to 44,1 kHz and it's multiples.
Then I realised, that the resampling doesn't affect the output of AirPlay content, which is really annoying.
The result is a badly stuttering audio playback (1 second playing as it should, 2 s break, 1 s playing...).

So the question is: what is the easiest way to resample the content from the shairport-sync output as well ?

Best regards, Markus.

asvedulf, 2022-01-16:
Any news on this?

timothyjward, 2022-01-16:
As of Volumio 3 shairport audio is now treated the same way as other audio. If you write a simple Volumio plugin to force 48kHz resampling then it should work for all audio sources including Shairport. 

There is documentation about how to plug into AAMPP [here](https://volumio.github.io/docs/Plugin_System/Custom_ALSA_Contributions.html)

asvedulf, 2022-01-17:
Ok, might need some help here.

Should I disable the Settings -> Playback Options -> Audio Resampling?

I have  created a plugin with has_alsa_contribution: true.
File with below contents -> plugin_dir/asound/resampler.Slave.conf:

```
pcm.resampler {
        type plug
        slave {
                pcm "Slave"
                format S24_LE
                rate 48000
        }
}
pcm.!default resampler
```

Double checked that the plugin is loaded in dev mode in live log.
Name of Soundcard is "Slave" referring to Card 1, Subdevice 0.
Have RPI Zero W connected to ADAU1701 over I2S. Using the internal clocks of the ADAU1701.

timothyjward, 2022-01-17:
> Should I disable the Settings -> Playback Options -> Audio Resampling?

It’s not required that you do, but it’s recommended. If you put different settings in your audio will get resampled twice.

> Name of Soundcard is "Slave" referring to Card 1, Subdevice 0.

Don’t target the sound card. As per the documentation you should target a pcm name that doesn’t exist so that Volumio can create it for you and join your contribution to the next stage in the AAMPP. 

> Have RPI Zero W connected to ADAU1701 over I2S. Using the internal clocks of the ADAU1701.
Output sound is distorted.

Can you see your contribution in /etc/asound.conf?

If so, Is it distorted with all audio, including native 48kHz? If so it’s likely a driver/hw issue. If it works fine with a 48kHz input file then it will be the resampling. 

You should be able to see what rate is being used with your device by using `cat` to display the audio device information under `/proc`

volumio, 2022-01-18:
@timothyjward @mervinderuiter 

This global resampling feature should be part of core Volumio. Who can have a look at it?

asvedulf, 2022-01-22:
Got it working according to [AAMPP](https://volumio.github.io/docs/Plugin_System/Custom_ALSA_Contributions.html) with below snippet. /proc device shows 48000 Hz. /etc/asound.conf adds an entry as well.

```
pcm.adau_in {
        type plug
        slave {
                pcm "adau_out"
                format S24_LE
                rate 48000
        }
}
```

Sound quality is better than comparing with my separate bluetooth dongle.

But agree with above, this alsa-configuration possibility should be part of core system.