volumio/Volumio2 #1552: Official Spotify Plugin: "My Playlists" is empty.

moritz1000, 2018-05-25:
I installed Volumio and only the official Spotify Plugin. When I go to Browse > Spotify the following playlists are working:
**Featured Playlists
What's New
Genres & Moods**

**My Playlists** is NOT working. When I click on it I just see a blank page with nothing on it. I deleted all my Spotify playlists but one. I tried another Spotify premium account... Nothing is working.

skikirkwood, 2018-05-26:
It just stopped working for me too.  It's a problem with the deprectated Libspotify.  To check, ssh into your Volumio device and type:

telnet localhost 6602

Now type:

ls

If you see something like this:

{
  "playlists" : [
    {
      "type" : "playlist",
      "name" : "",
      "tracks" : 0,
      "offline" : false,
      "index" : 0
    }
  ]
}

... then that verifies the old Libspotify-based app is not getting anything back from Spotify.

skikirkwood, 2018-05-26:
I turned logging on for the spop app and see that it is unable to login into Spotify with my credentials, which have been working fine for well over a year.  Spotify has been promising to such down access via Libspotify for years now.  Maybe they've finally done that.

volumio, 2018-05-27:
@skikirkwood Cant' test it as of now, hope that's not the case. Can you confirm when you can?

skikirkwood, 2018-05-27:
@volumio Yesterday Spotify stopped showing my personal playlists on all 3 of my Volumio systems.  All had been working fine before.  I tracked it down to Spop getting back a null result with the "ls" command:

Volumio Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
volumio@kef:~$ telnet localhost 6602
Trying 127.0.0.1...
Connected to localhost.
Escape character is '^]'.
spop 0.0.1
ls
{
  "playlists" : [
    {
      "type" : "playlist",
      "name" : "",
      "tracks" : 0,
      "offline" : false,
      "index" : 0
    }
  ]
}

And then after turning Spop logging on I saw an error message trying to login to Spotify.

I'm also seeing some chatter other distributions such as Mopidy that use Libspotify have suddenly stopped working this week.  https://discourse.mopidy.com/t/cant-load-spotify-playlists-recently/2522

I did install 2.403 on all systems yesterday before I tested Spotify, so there's some small chance there's a regression there, but I doubt it.

Looks to me like Spotify finally turned off API calls from Libspotify.

skikirkwood, 2018-05-27:
I can also confirm Balbuze's Spotify Connect2 is working fine on all 3 Volumios, a Raspberry Pi B+, Pi 3 and Pi 3 B+.

skikirkwood, 2018-05-27:
Also interesting is Spotify putting out a new"[Web Playback SDK](https://beta.developer.spotify.com/documentation/web-playback-sdk/) that runs on OS X/Windows/Linux. It makes use of Encrypted Media Extensions in browsers such as Chrome and FF to allow playback of Spotify within those browsers.

The API looks really trivial to use, so I wonder if the browser component that supports this could be added to Volumio, enabling full access/playback of Spotify content for premium users.  Still would have to handle the Oauth authentication step.

joeythesaint, 2018-05-28:
This has been happening for me for just about a week now, I noticed `My Playlists1 failing last Tuesday but I'm pretty sure it was working on Monday.  As @moritz1000 describes, the `Featured`, `Genres` and `New` lists are fine and I can play stuff from them successfully.  I can also use the spotify-connect2 plugin successfully to play things off my phone or other computer without issue.

I still have a valid use case for this plugin, though, as it allows me to access the music players in other rooms that are connected to other Spotify accounts.  There are four people in my house, all with their own accounts and so far we had been using our own volumio interfaces and sharing some playlists, but with this regression now I can't, for example, start music in two different rooms.

In regard to regression, I think it's already well understood, but just to make it perfectly clear, all of my volumio devices are still running 2.389, so it's not something new to 2.403, @skikirkwood, so there's no need to consider down-grading your devices.  :-)

volumio, 2018-05-28:
@skikirkwood what if we use the spotify web api to get also the playlists? would that work?

skikirkwood, 2018-05-28:
I think it would work @volumio. It looks like spop is still working for playback and control of Spotify tracks, only the "ls" command to get a user's playlists has stopped functioning.

I actually left a placedholder in the explodeuri function to replace the spop "ls" with the Spotify Web API call, line 1583 on the plugin's index.js file.  

So to try this out, we'd want to use the example app on Spotify Developer's web site to authenticate with Spotify and get back a access token and refresh token.  Then copy those tokens into the index.js file, and use the access token to make a Spotify Web API call to get a user's playlists. Replace the spop ls call in explodeuri with the new Spotify Web API code, and with some luck, it might actually work!

Getting this into production, we'd need to think of some clever way for people to be able to do the authentication with their Spotify account, and get the access and refresh tokens into their Volumio system.

volumio, 2018-05-28:
@clever! Let's try this way, after we can think also at oauth...

volumio, 2018-05-28:
Thanks Jim!

skikirkwood, 2018-05-28:
As you can see, others using Libspotify had the same result as us this week: https://github.com/mopidy/mopidy-spotify/issues/182

skikirkwood, 2018-05-29:
[image]
Making progress @volumio - now let's see if we can actually play the tracks on my playlists!

skikirkwood, 2018-05-29:
[image]
Damn - it works! 

Ok, to get this to usable we need to figure out a way to get the Oauth access and refresh token for each user into their Volumio system. Do something about pagination since it's defaulting to the first 20 playlists.  And that's about it.

@volumio I'll park the current prototype in Github somewhere and send you a link to it.

skikirkwood, 2018-05-30:
You can grab the work-in-progress file here @volumio - https://github.com/skikirkwood/volumio-plugins/blob/Spotify2/plugins/music_service/spotify/index.js

volumio, 2018-05-30:
@skikirkwood well done!!!! Great news! I won't be able to do much the next 2 days.. but if there is something urgent where I can help, tell me . good one!

skikirkwood, 2018-05-30:
@volumio what I'll do is try to hack something up where I extend the Spotify credentials form with spots for the access and refresh token.  Then I'll put the check in for the expired access token and grab a new one with the refresh token.  

With that working, I'll put up a temporary site (e.g. volumio.jobflo.com) on my startup's hosting provider where people can go, authenticate with Spotify and get access/refresh tokens.  They can then copy and paste them into the Spotify credentials form, and while it's a hack, it should get us working again.

The great news is the UI looks greatly improved using the playlist artwork from Spotify.  Plus, it will now be trivial to add links for "My Albums", "My Tracks", "My Favorites", etc. The Spotify Web API even has a function to generate playlists based upon seeds of genre's, artists and tracks, so we could have a lot of fun here.

skikirkwood, 2018-05-31:
Ok, good news is the refresh token works, so we definitely have a working system.  Oauth gives you an access token and a refresh token, but we only need the refresh token.  I use the refresh token at startup to get a new access token, and then grab a new access token using the refresh token after one hour.

So @volumio, let's figure out how to make this usable for the Volumio community and we can get this out in a new release.  In the mean time let me see if I can add "My Albums", "My Tracks", etc.

skikirkwood, 2018-05-31:
I could use some help testing the updated plugin - here's how you can get access to it:

https://volumio.org/forum/help-needed-test-spotify-plugin-with-oauth-t10071.html

volumio, 2018-06-01:
@skikirkwood I have an idea... What about having a call monday to discuss about it?

skikirkwood, 2018-06-01:
Sure, I am tied up at work until 5pm California time.  I have a Zoom account we could use for a video conference if you want.

NellyWhadsDev, 2018-06-18:
@volumio @skikirkwood Any updates on this?

antonsn, 2018-10-14:
still not working, I know alexa app can directly open spotify app and integrate it, would be great to see similar apporoach in spotify plugin

damien122, 2018-10-21:
@skikirkwood your fix works fine. When do you plan to merge it to the main branche an create a new release version?

skikirkwood, 2018-10-21:
Hi @damien122 , I need to integrate some changes another contributor made so you can just plug the Spotify refresh token into your Volumio credentials screen and not have to edit the modified index.js file and copy it to your Volumio device.  I'd also like to add pagination of some kind to get past the 50 item API cap for results.

Hope to do this soon, just haven't had any time recently.

damien122, 2018-10-28:
@skikirkwood sounds good 👍 could you please notify us here when you release the new version, so we can test it.

LunkRat, 2019-01-26:
How can I help get this committed? 

> Ok, good news is the refresh token works, so we definitely have a working system. Oauth gives you an access token and a refresh token, but we only need the refresh token. I use the refresh token at [...]

gmasonxoom, 2019-02-05:
I would like to get this functionality working too, if there's anything I can do to test or assist in some other way then please let me know.

damien122, 2019-03-28:
Hi, after I modified the plugin settings it was not possible for me to get an volumio update. I was forced to to a factory reset and then install the update. But now the spotify plugin can't play any song. It just shows me some featured lists and some more but it can't play them. What's happen there? Any one an idea?

bqstony, 2019-09-03:
@skikirkwood, @majko96 @volumio any plans for update this fix to a new offical plugin version?

GrandNewbien, 2019-10-10:
Looking for alternative to volumio now. The hacky fixes aren't the most robust thing to get around.

volumio, 2019-10-10:
It's on our TODO List, we need to set up the OAUTH flow for Spotify to get working again.
Unfortunately we always have other priorities that supersede this... But rest assured we'll make it happen as soon as we can

HermitHash, 2020-03-01:
@volumio - what Volumio release is the "My Playlist" fixed in? I'm currently running 2.703 and still have the issue.

Love Volumio!!

NellyWhadsDev, 2020-03-01:
Don't think this was ever merged; still have to install this manually to make it happen.

HermitHash, 2020-03-01:
Thanks @NellyWhadsDev. 

I should have checked the Volumio2 Plugins page before I posted. Installed the Spotify plugin v2.0.2, after install it took me to the Spotify Authorize page, after providing my credentials, 'My Playlists' works.

NellyWhadsDev, 2020-03-02:
Blessed. I'll have that a try tonight!