volumio/Volumio2 #1559: Darn private playlists no longer showed

edegraaff, 2018-06-03:
Hi, i am using Volumio, with the spotify plugin for quite some time, and since one moment in time my playlists are not longer visible. Could you look into this? if needed logging and other info will be supplied.

regards

eelco

skikirkwood, 2018-06-03:
https://volumio.org/forum/help-needed-test-spotify-plugin-with-oauth-t10071.html

edegraaff, 2018-06-03:
Hi did the patching, and it worked partly. I have one entry in my playlist so it worked a bit. I also refeshed the token and added it again, with same result. 
[image]

regards

eelco

skikirkwood, 2018-06-03:
Hi @edegraaff , I just updated the Oauth server to give two additional authorization scopes.  Can you get a new refresh token and see if that shows more of your playlists.  Thanks!

skikirkwood, 2018-06-03:
Could you post your /var/log/volumio.log file here.  Unfortunately these things are a bit challenging to debug!

edegraaff, 2018-06-03:
[volumio.log](https://github.com/volumio/Volumio2/files/2066299/volumio.log)
Here you are ... yes i can imagine hope it helps you

skikirkwood, 2018-06-03:
Thanks for that.  Odd though, I can't find any log event of listing your playlists.  Can you copy to your volumio server this latest version of the Spotify plugin's index.js file into /data/plugins/music_service/spop after getting a new refresh token and try it again, and post the log file again.  Thanks for your help!

https://github.com/skikirkwood/volumio-plugins/blob/myalbums/plugins/music_service/spotify/index.js

edegraaff, 2018-06-03:
Done and guess.... it works ! 
the Token variable looks to to be replaced to line 451 btw
i am new to github, if i want to download a file like the index.js is that possible?  i did now cut and paste.
i installed also the githup desktop software.

skikirkwood, 2018-06-03:
Awesome!  Yes, forgot that the refresh token is in a different place in the newest version.

If you go to the top level of any Github repo you are looking at, for example here:

https://github.com/skikirkwood/volumio-plugins

In the top right you have the green "Clone or Download" button which will give you an option to download a Zip file of the entire repo.

edegraaff, 2018-06-03:
I will try to learn here also, thanks for your help to fix this issue.

mydudechris, 2019-06-26:
Just installed the latest version and got it running on my Rpi3+. Everything seems to work except the playlists feature, none of my playlists show up. Public or private. I'm on a family plan, would that have anything to do with it? Thanks @skikirkwood

mydudechris, 2019-06-26:
The above fixed the issue for me. Do I need keep providing http://54.86.144.136 with access? @skikirkwood