volumio/Volumio2 #1560: Toggle mute/unmute possible

Saiyato, 2018-06-13:
Hi there,

I thought there once was the option to toggle mute/unmute, this is not there anymore.
Did it break stuff and/or is it possible to bring it back? I can work around it a bit by changing the volume, but it feels a bit dirty.

Thanks a million

I believe the command was: `volumio volume toggle`

jlagden, 2018-06-19:
I noticed this when I was doing my remote control configuration last year. I'm sure the command was in the documentation but it's not there now.<br><br>
In the end I just ignored it as I found doing a `volumio volume plus` or `minus` would unmute, so I could use the relevant buttons on the remote to do an unmute.<br><br>

However `volumio volume toggle` does do something if you look at `/var/log/volumio.log` (Calls mute/unmute depending on the current volume state)
<br><br>

The only thing I don't understand is when I call toggle the WebUI doesn't update, but it does if I call mute or unmute.<br><br>

I can't test further until tonight as I'm not in front of the box.

Saiyato, 2018-06-20:
Oh wow, hadn't even noticed the entries in the log... but yes, once upon a time it did work. In the sense that it actually toggled the volume.

I'm seeing the following, so I might be able to patch it:
```
Jun 20 10:27:10 volumioboss volumio[790]: info:
Jun 20 10:27:10 volumioboss volumio[790]: ---------------------------- Client requests Volume toggle
Jun 20 10:27:10 volumioboss volumio[790]: info: VolumeController::SetAlsaVolumetoggle
Jun 20 10:27:10 volumioboss volumio[790]: info: VolumeController::SetAlsaVolumemute
Jun 20 10:27:10 volumioboss volumio[790]: null
Jun 20 10:27:12 volumioboss volumio[790]: info:
Jun 20 10:27:12 volumioboss volumio[790]: ---------------------------- Client requests Volume toggle
Jun 20 10:27:12 volumioboss volumio[790]: info: VolumeController::SetAlsaVolumetoggle
Jun 20 10:27:12 volumioboss volumio[790]: info: VolumeController::SetAlsaVolumeunmute
Jun 20 10:27:12 volumioboss volumio[790]: null
Jun 20 10:27:12 volumioboss volumio[790]: info: VolumeController::Volume unmute
```

Note that it mutes first and then unmutes, so it actually does 'something'

If we compare this to muting (which does reflect in the GUI).
```
Jun 20 10:28:45 volumioboss volumio[790]: info:
Jun 20 10:28:45 volumioboss volumio[790]: ---------------------------- Client requests Volume mute
Jun 20 10:28:45 volumioboss volumio[790]: info: VolumeController::SetAlsaVolumemute
Jun 20 10:28:45 volumioboss volumio[790]: null
Jun 20 10:28:45 volumioboss volumio[790]: info: CoreStateMachine::pushState
Jun 20 10:28:45 volumioboss volumio[790]: info: CoreStateMachine::getState
Jun 20 10:28:45 volumioboss volumio[790]: info: CorePlayQueue::getTrack 0
Jun 20 10:28:45 volumioboss volumio[790]: info: CoreCommandRouter::volumioPushState
Jun 20 10:28:45 volumioboss volumio[790]: info: CoreCommandRouter::executeOnPlugin: volumiodiscovery , saveDeviceInfo
Jun 20 10:28:45 volumioboss volumio[790]: info: interfaceApi::pushState
Jun 20 10:28:45 volumioboss volumio[790]: info: Pushing Favourites {"service":"mpd","uri":"mnt/NAS/HD/Dire Straits - Brothers in Arms/02 Money For Nothing.flac","favourite":false}
```

Or volume up for that matter:
```
Jun 20 10:29:50 volumioboss volumio[790]: info:
Jun 20 10:29:50 volumioboss volumio[790]: ---------------------------- Client requests Volume +
Jun 20 10:29:50 volumioboss volumio[790]: info: VolumeController::SetAlsaVolume+
Jun 20 10:29:50 volumioboss volumio[790]: null
Jun 20 10:29:50 volumioboss volumio[790]: info: VolumeController::Volume 0
Jun 20 10:29:50 volumioboss volumio[790]: info: CoreStateMachine::pushState
Jun 20 10:29:50 volumioboss volumio[790]: info: CoreStateMachine::getState
Jun 20 10:29:50 volumioboss volumio[790]: info: CorePlayQueue::getTrack 0
Jun 20 10:29:50 volumioboss volumio[790]: info: CoreCommandRouter::volumioPushState
Jun 20 10:29:50 volumioboss volumio[790]: info: CoreCommandRouter::executeOnPlugin: volumiodiscovery , saveDeviceInfo
Jun 20 10:29:50 volumioboss volumio[790]: info: interfaceApi::pushState
Jun 20 10:29:50 volumioboss volumio[790]: info: Pushing Favourites {"service":"mpd","uri":"mnt/NAS/HD/Dire Straits - Brothers in Arms/02 Money For Nothing.flac","favourite":false}
```

One of the big difference that I notice, is that the mute and volume plus command trigger a 'pushstate', whereas volume toggle doesn't. I'm pretty sure it doesn't actually update ALSA, but I'll have to check that tonight I guess.

Saiyato, 2018-06-20:
Ok, I found the problem, the promise was not propagated properly, will PR in a minute.

jlagden, 2018-06-20:
Nice! Thanks 👍

Saiyato, 2018-06-21:
yw, needed this fix myself too ;)

PR #1564 (2018-06-25): Fixed the volume toggle function
The promise was not propagated properly, this should work again now.

Plus updated some comments, minor grammatical changes.

fixes #1560
commit 77736b82 (2018-06-25): Fixed the volume toggle function (#1564)

The promise was not propagated properly, this should work again now.

Plus updated some comments, minor grammatical changes.

fixes #1560
closed 2018-06-21 by Saiyato