volumio/Volumio2 #1567: After update Volumio on Berryboot system failed to boot

edegraaff, 2018-06-22:
After the update of Volumio from the Volumio webinterface the system crashed. 
Tried to repair it by dooing a new curl of Volumio and reinstalled the whole.
Berryboot boot > Volumio fine but the webinterface did not came up.
By using manual boot, there seemed to be an problem in the index/js script since i see a fail here.
What is de cause and how to i fix this one? ?

thanks

Error: bind EADDRINUSE null:3001
    at Object._errnoException (util.js:1022:11)
    at _exceptionWithHostPort (util.js:1044:20)
    at listenOnMasterHandle (net.js:1427:16)
    at rr (internal/cluster/child.js:121:12)
    at Worker.send (internal/cluster/child.js:88:7)
    at process.onInternalMessage (internal/cluster/utils.js:42:8)
    at emitTwo (events.js:131:20)
    at process.emit (events.js:214:7)
    at emit (internal/child_process.js:772:12)
    at _combinedTickCallback (internal/process/next_tick.js:141:11)
Worker 9 died :(

volumio, 2018-06-22:
No idea, and it's quite hard to have a solution with the stack trace you provided.
Volumio is meant to run in its environment, and berryboot is not the "official" way there, so sorry but can't help...

edegraaff, 2018-06-22:
ok no problem i make a clean install it is not that hard.

edegraaff, 2018-06-23:
Clean installation using the curl -Ls --output installVolumio2 http://bit.do/InstV2; chmod u+x installVolumio2; ./installVolumio2

It did not do the job... seems there is something broke but i can't find out where. The os is booting fine, the issue seems to be in the webservices i presume it is the same issue i detected after the update from the webgui.

reference:
https://volumio.org/forum/multiboot-volumio2-with-kodi-under-berryboot-t6818.html

i am willing to debug but need some help here.

volumio, 2018-06-23:
This is not a clean install, but a totally unsupported way of doing things. Please ask to the one which posted the guide and created the script (which is already supporting you)

closed 2018-06-23 by volumio