volumio/Volumio2 #1568: System Update does not preserve /boot/config.txt changes

swinokur, 2018-06-22:
Hi,
It looks like (based on the 2 recent system updates) that changes to /boot/config.txt aren't being preserved across the update.

config.txt is copied to /imgpart/config.txt.bak during the upgrade process, but across the system restart (post-upgrade), a 'vanilla' config.txt is placed in /boot.

My /boot/config.txt pre-upgrade:
```
root@volumio:/boot# ls -l config.txt.save
-rw-rw-rw- 1 root root 229 Jun 22 19:35 config.txt.save
```

```
root@volumio:/boot# cat config.txt.save
initramfs volumio.initrd
gpu_mem=16
max_usb_current=1
dtparam=audio=on
audio_pwm_mode=2
dtparam=i2c_arm=on
disable_splash=1
hdmi_force_hotplug=1
disable_overscan=1
hdmi_group=1
hdmi_mode=16
hdmi_ignore_cec_init=1
dtparam=eee=off
```

and after upgrade:
```
root@volumio:/boot# ls -l config.txt
-rw-rw-rw- 1 root root 147 Jan  1  1980 config.txt
```
```
root@volumio:/boot# cat config.txt
initramfs volumio.initrd
gpu_mem=16
max_usb_current=1
dtparam=audio=on
audio_pwm_mode=2
dtparam=i2c_arm=on
disable_splash=1
hdmi_force_hotplug=1
```

thanks!

swinokur, 2018-07-26:
Hi,
This issue still exists in the latest update.

xipmix, 2018-07-30:
I think this is happening in https://github.com/volumio/Build/blob/master/scripts/init  or https://github.com/volumio/Build/blob/master/scripts/raspberryconfig.sh. I'll open an issue there pointing here.

Can you please confirm which platform you are on - I am assuming RPi.

swinokur, 2018-07-30:
```
volumio@volumio:~$ cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
VOLUMIO_BUILD_VERSION="b38474a34dd6501a9ca2801645a6a861fdbc549d"
VOLUMIO_FE_VERSION="f027ce75738c64189018057558e04cba7a74a080"
VOLUMIO_BE_VERSION="70b3a2f172874ee2df1649bd7da540b026dbcc47"
VOLUMIO_ARCH="arm"
VOLUMIO_VARIANT="volumio"
VOLUMIO_TEST="FALSE"
VOLUMIO_BUILD_DATE="Thu Jul 26 10:29:04 CEST 2018"

VOLUMIO_VERSION="2.443"
VOLUMIO_HARDWARE="pi"
VOLUMIO_HASH="2b389f6d4708994da17211c23c08c406"
```