volumio/Volumio2 #1575: CLI command vrestart only stops when called from within Volumio

Saiyato, 2018-07-02:
If you call the vrestart command in the CLI from a plugin only the stop is registered.
I'm not sure if this is by design, but this means you need to add socket IO for a plugin to communicate to Volumio.

Example:
```
	exec(command, {uid:1000, gid:1000}, function (error, stout, stderr) {		
		if(error)
			self.logger.error(stderr);
		
		defer.resolve();
	});
```

where command = `/usr/local/bin/volumio vrestart`

volumio, 2018-07-03:
Why do you need to do it? There are for sure better ways to do it

Saiyato, 2018-07-03:
I want to be able to restart Volumio from within my plugin without having to add all the node modules for a socket io call.

My use case is that I test a lot and I want to be able to restart Volumio using a long press of my rotary encoder. For all other functions I could use the CLI or the command router, which doesn't need any additional node module (and its dependencies).

If I call the vrestart from an SSH session it works, but not if I execute the command using exec.

volumio, 2018-07-03:
Do you mean restart the system or restart volumio process itself?
For the first, call the command router restart method. For the latter execute with exec
/usr/bin/sudo /bin/killall node 

but handle with great care!

Saiyato, 2018-07-03:
I mean just the Volumio process and I'm trying to avoid any sudo calls. Is there a preference for killall node as opposed to systemctl restart volumio? Normally I use the latter.

Naturally I'll use it with care, to avoid wear and tear on the SD card. :)