volumio/Volumio2 #1579: Empty album view

murfblurp, 2018-07-06:
Hi,

I installed Volumio on a 32GB card used on a Pi 3B and linked it to my music library stored on a NAS.

My library is consisting of 4460 albums (87785 tracks) of flac and dsd.

When looking for a specific album I can retrieve it and play it without issue. 

View by genre is working for the genre in which I don't have many albums but not showing anything for the bigger ones. Album view without any filter is not showing me anything.

Is there any library size limitation or a cache size which can be increased in order to solve my issue?

Thanks.

shrug, 2018-07-09:
I have the same issue, although my library is somewhat smaller - 2197 albums, 35585 tracks. A mix of (mostly) FLAC and MP3. Also stored on a NAS. Volumio 2.413 on a Pi 3. 

I have not had the chance to do much troubleshooting yet.

shrug, 2018-07-11:
I get the following log entries when browsing Albums

```
Jul 11 04:05:38 volumio volumio[1032]: info: CoreCommandRouter::executeOnPlugin: mpd , handleBrowseUri
Jul 11 04:05:38 volumio volumio[1032]: info: CURURI: albums://
Jul 11 04:05:39 volumio volumio[1032]: info: CoreCommandRouter::volumioGetState
Jul 11 04:05:39 volumio volumio[1032]: info: CoreStateMachine::getState
Jul 11 04:05:39 volumio volumio[1032]: info: CorePlayQueue::getTrack 0
Jul 11 04:05:39 volumio volumio[1032]: info: CoreCommandRouter::volumioGetState
Jul 11 04:05:39 volumio volumio[1032]: info: CoreStateMachine::getState
Jul 11 04:05:39 volumio volumio[1032]: info: CorePlayQueue::getTrack 0
Jul 11 04:05:40 volumio volumio[1032]: error: MPD error: Error: read ECONNRESET
```

volumio, 2018-07-11:
Seems your mpd is crashing. Please provide a full log:
https://volumio.github.io/docs/User_Manual/Sending_logs_for_troubleshooting.html

streetdiver, 2018-08-15:
I also have the same issue: volumio 2.143 and earlier versions, Raspberry 2, external HDD (no NAS), about 4000 albums, 90000 tracks.
I assume it's a capacity problem: I have a big folder A with mp3 and flac files. These files are in two subfolders "classic" and "other" with almost the same size. When I scan folder A including "classic" and "other", the browser tabs wheel did not stop spinning. I waited for 24 hours and then rebooted and the album view was empty.
When I used .mpdignore in folder "classic" then only this folder was scanned, the wheel stopped after some time and the album's view was ok! 
When I used .mpdignore in folder "other" then only this folder was scanned, the wheel stopped after some time and the album's view was again ok! 
When I first applied .mpdignore in one of these folders scanned the other folder and then removed the .mpdignore and scanned this directory also, the spinning wheel stopped and the album's view was ok for the whole folder A! But after some time, only playing tracks but not adding or removing tracks, the album view was empty again!?
I did these tests after increasing a buffer size in mpdconf from 16.384 or so to 32.768. (Sorry I cannot check the precise buffer name and cannot send a log at the moment. The message was also "MPD error: Error: read ECONNRESET")
I would be very glad if my findings help to remove this problem!

streetdiver, 2018-08-30:
Hi,
Installed version 2.452 (without any modifications) with the same result. I sent a log file dk2AI7C.html.
Regards

xipmix, 2018-09-01:
I looked at https://logs.volumio.org/volumio/dk2AI7C.html but I get a connection timeout.

streetdiver, 2018-09-01:
Very sorry for not posting the complete URL!
https seems to be the problem: please try http://logs.volumio.org/volumio/dk2Al7C.html 
Regards

streetdiver, 2018-09-01:
I increased max_buffer_size from about 16.500 to 116.500 and it seems to work in the current version of volumio 2.452! I couldn't do this in earlier versions! Great!
I will continue to test this!
Regards