volumio/Volumio2 #1582: Volume on REST Api

Azalius, 2018-07-13:
Hi all,
I'm having a problem while trying to change the volume with the REST api.
The very command of the documentation isn't working (https://volumio.github.io/docs/API/REST_API.html).
I tried (ssh on rpi 3 - latest version of Volumio)
`wget -O/dev/null http://localhost:3000/api/v1/commands?cmd=volume&volume=80`
That works for other commands but when I try this one the WebUi change its volume to "NaN" then to zero.  Its the same for plus - minus - mute - unmute.
Does someone else have the same problem / can try it? 
Is my syntax wrong ? If that's the case maybe update the doc that appear to be false.
Thanks for your help in advance.

micdudk, 2018-09-18:
I have the same problem, when trying to change volume using REST api it is always set to NaN/null.

log:
`---------------------------- Client requests Volume NaN
2018-09-18T12:43:05.806Z - info: VolumeController::SetAlsaVolumeNaN
2018-09-18T12:43:05.844Z - info: VolumeController::Volume NaN
2018-09-18T12:43:05.846Z - info: CoreStateMachine::pushState
2018-09-18T12:43:05.847Z - info: CoreStateMachine::getState
2018-09-18T12:43:05.848Z - info: CorePlayQueue::getTrack 0
2018-09-18T12:43:05.848Z - info: CoreCommandRouter::volumioPushState
2018-09-18T12:43:05.849Z - info: CoreCommandRouter::executeOnPlugin: volumiodiscovery , saveDeviceInfo
2018-09-18T12:45:59.789Z - info: 
`
I have v2.457

richardklingler, 2019-10-30:
Hello

Will this never be fixed?
Still no volume control via HTTP REST API...

volumio, 2019-10-30:
@richardklingler I just tested it and it just works...
Make sure you send a GET request and that the whole URL is enclosed in " .

For example (this works):
curl -X GET "http://localhost:3000/api/v1/commands?cmd=volume&volume=70"

closed 2019-10-30 by volumio