volumio/Volumio2 #1592: Impossible to install Chrome or Firefox

biva, 2018-08-14:
I use the Kodi add-on, which is great (thank you again!). And I'd like to have a browser under Kodi: I understood that I need to use the `chome launcher` Kodi plugin, that I installed. But then, I need to install either Firefox or Chrome on my RPI3:
`sudo apt-get install firefox (or chromium-browser)`

The result is the following:
<code>Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 chromium-browser : Depends: libraspberrypi0 but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
</code>

Any idea to solve this? Thank you!

biva, 2019-02-23:
No idea?

macmpi, 2019-02-26:
`libraspberrypi0` and al. are packages which change Pi firmware files on boot partition, and are therefore forbidden to install on Volumio distribution. Some [protection mesures](https://github.com/volumio/Build/blob/master/scripts/raspberryconfig.sh#L121-L131) have been undertaken, and that that's what you are seeing.
[touch_display](https://github.com/volumio/volumio-plugins/blob/master/plugins/miscellanea/touch_display/install.sh) plugin install procedure (which installs chrome) takes care about that: you may want to look at it to manually install your missing pieces (or install that plugin overall to get chromium).

biva, 2019-02-26:
Thank you @macmpi , i'll look into this :)

biva, 2019-02-27:
Thanks @macmpi 
I installed the full touch_display plugin, but Kodi cannot find Chrome. I see an option `Use custom Chrome/script path`. Should I specify a custom path? Which one?
Thanks!

macmpi, 2019-02-27:
No idea: there's probbaly some custom Kodi tweak to set.
Maybe re-installing plugin on Kodi side might do it for you, now binaries are available...

I run Kodi on its own, separately from Volumio2 via [dual-boot (berryboot)](https://volumio.org/forum/multiboot-volumio2-with-kodi-under-berryboot-t6818.html) on same SD: I don't get into such kind of mixed setup issues as each distribution has it's own environement.

biva, 2019-03-01:
Thanks @macmpi 
Do you know the path where the plugin touch_display installs Chrome?

macmpi, 2019-03-01:
probably in `/usr/bin/` something.
try `ls /usr/bin/chromium*`  or `find /usr -name chromium*`

biva, 2019-03-01:
Unfortunately, I can't find it. `sudo find -name chromium` does not find anything either :(

I verified, and the plugin TouchDisplay 1.0.0 is installed. How can I check if chromium is really installed?

biva, 2019-03-01:
My log is:

2019-03-01T10:24:45.671Z - info: You might want to run 'apt-get -f install' to correct these.
2019-03-01T10:24:45.675Z - info: The following packages have unmet dependencies:
2019-03-01T10:24:45.684Z - info:  libc6 : Depends: libgcc1 but it is not installed
2019-03-01T10:24:45.710Z - info: Installing Graphical environment
2019-03-01T10:24:45.769Z - info: Reading package lists...
2019-03-01T10:24:46.422Z - info: Building dependency tree...
2019-03-01T10:24:46.425Z - info: Reading state information...
2019-03-01T10:24:46.711Z - info: You might want to run 'apt-get -f install' to correct these:
2019-03-01T10:24:46.715Z - info: The following packages have unmet dependencies:
2019-03-01T10:24:46.751Z - info:  libc6 : Depends: libgcc1 but it is not going to be installed
2019-03-01T10:24:46.754Z - info:  openbox : Depends: libglib2.0-0 (>= 2.35.9) but it is not going to be installed
2019-03-01T10:24:46.755Z - info:            Depends: libice6 (>= 1:1.0.0) but it is not going to be installed
2019-03-01T10:24:46.757Z - info:            Depends: libobrender29 (>= 3.5.2) but it is not going to be installed
2019-03-01T10:24:46.758Z - info:            Depends: libobt2 (>= 3.5.2) but it is not going to be installed
2019-03-01T10:24:46.760Z - info:            Depends: libsm6 but it is not going to be installed
2019-03-01T10:24:46.761Z - info:            Depends: libstartup-notification0 (>= 0.7) but it is not going to be installed
2019-03-01T10:24:46.762Z - info:            Depends: libx11-6 but it is not going to be installed
2019-03-01T10:24:46.764Z - info:            Depends: libxau6 but it is not going to be installed
2019-03-01T10:24:46.765Z - info:            Depends: libxcursor1 (> 1.1.2) but it is not going to be installed
2019-03-01T10:24:46.766Z - info:            Depends: libxext6 but it is not going to be installed
2019-03-01T10:24:46.768Z - info:            Depends: libxi6 (>= 2:1.2.99.4) but it is not going to be installed
2019-03-01T10:24:46.769Z - info:            Depends: libxinerama1 but it is not going to be installed
2019-03-01T10:24:46.771Z - info:            Depends: libxrandr2 but it is not going to be installed
2019-03-01T10:24:46.772Z - info:            Recommends: obconf but it is not going to be installed
2019-03-01T10:24:46.773Z - info:            Recommends: python-xdg but it is not going to be installed
2019-03-01T10:24:46.775Z - info:            Recommends: scrot but it is not going to be installed
2019-03-01T10:24:46.776Z - info:  xinit : Depends: libx11-6 but it is not going to be installed
2019-03-01T10:24:46.778Z - info:          Depends: coreutils (>= 7.4-1) but it is not going to be installed
2019-03-01T10:24:46.779Z - info:          Depends: x11-common but it is not going to be installed
2019-03-01T10:24:46.780Z - info:          Depends: xauth but it is not going to be installed
2019-03-01T10:24:46.782Z - info:          Recommends: xserver-xorg but it is not going to be installed or
2019-03-01T10:24:46.783Z - info:                      xserver
2019-03-01T10:24:46.785Z - info:  xorg : Depends: xserver-xorg (>= 1:7.7+7+b1) but it is not going to be installed
2019-03-01T10:24:46.786Z - info:         Depends: libgl1-mesa-glx but it is not going to be installed or
2019-03-01T10:24:46.788Z - info:                  libgl1
2019-03-01T10:24:46.789Z - info:         Depends: libgl1-mesa-dri but it is not going to be installed
2019-03-01T10:24:46.790Z - info:         Depends: libglu1-mesa but it is not going to be installed
2019-03-01T10:24:46.792Z - info:         Depends: xfonts-base (>= 1:1.0.0-1) but it is not going to be installed
2019-03-01T10:24:46.793Z - info:         Depends: xfonts-100dpi (>= 1:1.0.0-1) but it is not going to be installed
2019-03-01T10:24:46.795Z - info:         Depends: xfonts-75dpi (>= 1:1.0.0-1) but it is not going to be installed
2019-03-01T10:24:46.796Z - info:         Depends: xfonts-scalable (>= 1:1.0.0-1) but it is not going to be installed
2019-03-01T10:24:46.798Z - info:         Depends: x11-apps but it is not going to be installed
2019-03-01T10:24:46.800Z - info:         Depends: x11-session-utils but it is not going to be installed
2019-03-01T10:24:46.801Z - info:         Depends: x11-utils but it is not going to be installed
2019-03-01T10:24:46.803Z - info:         Depends: x11-xkb-utils but it is not going to be installed
2019-03-01T10:24:46.805Z - info:         Depends: x11-xserver-utils but it is not going to be installed
2019-03-01T10:24:46.806Z - info:         Depends: xauth but it is not going to be installed
2019-03-01T10:24:46.814Z - info:         Depends: xfonts-utils but it is not going to be installed
2019-03-01T10:24:46.815Z - info:         Depends: xkb-data but it is not going to be installed
2019-03-01T10:24:46.817Z - info:         Depends: xorg-docs-core but it is not going to be installed
2019-03-01T10:24:46.818Z - info:         Depends: xterm but it is not going to be installed or
2019-03-01T10:24:46.820Z - info:                  x-terminal-emulator
2019-03-01T10:24:46.868Z - info: Installing Chromium
2019-03-01T10:24:46.949Z - info: Reading package lists...
2019-03-01T10:24:47.685Z - info: Building dependency tree...
2019-03-01T10:24:47.690Z - info: Reading state information...
2019-03-01T10:24:47.969Z - info: You might want to run 'apt-get -f install' to correct these:
2019-03-01T10:24:47.973Z - info: The following packages have unmet dependencies:
[... cut ...]
2019-03-01T10:24:48.281Z - info: Install script completed
2019-03-01T10:24:48.284Z - info: Adding reference to registry
2019-03-01T10:24:48.288Z - info: Done installing plugin.
2019-03-01T10:24:48.334Z - info: Folder /tmp/plugins removed
2019-03-01T10:24:48.334Z - info: Folder /tmp/downloaded_plugin.zip removed
2019-03-01T10:24:48.338Z - info:   Allowing volumio to start an xsession
2019-03-01T10:24:48.340Z - info: Disabling Kiosk Service
2019-03-01T10:24:48.359Z - info: Folder /data/temp removed

biva, 2019-03-01:
And if I do manually the steps in install.sh, I have some errors:

debconf: delaying package configuration, since apt-utils is not installed
Selecting previously unselected package gcc-4.9-base:armhf.
(Reading database ... 48 files and directories currently installed.)
Preparing to unpack .../gcc-4.9-base_4.9.2-10+deb8u2_armhf.deb ...
Unpacking gcc-4.9-base:armhf (4.9.2-10+deb8u2) ...
Setting up gcc-4.9-base:armhf (4.9.2-10+deb8u2) ...
Selecting previously unselected package multiarch-support.
(Reading database ... 58 files and directories currently installed.)
Preparing to unpack .../multiarch-support_2.19-18+deb8u10_armhf.deb ...
Unpacking multiarch-support (2.19-18+deb8u10) ...
Selecting previously unselected package libc6:armhf.
Preparing to unpack .../libc6_2.19-18+deb8u10_armhf.deb ...
Unpacking libc6:armhf (2.19-18+deb8u10) ...
Selecting previously unselected package libgcc1:armhf.
dpkg: regarding .../libgcc1_1%3a4.9.2-10+deb8u2_armhf.deb containing libgcc1:armhf, pre-dependency problem:
 libgcc1 pre-depends on multiarch-support
  multiarch-support is unpacked, but has never been configured.

dpkg: error processing archive /var/cache/apt/archives/libgcc1_1%3a4.9.2-10+deb8u2_armhf.deb (--unpack):
 pre-dependency problem - not installing libgcc1:armhf
Errors were encountered while processing:
 /var/cache/apt/archives/libgcc1_1%3a4.9.2-10+deb8u2_armhf.deb

macmpi, 2019-03-01:
(you probably need to `sudo find`)
Is the plugin working at all? This would definitely give you clear indication if all is installed properly :)

It must be [/usr/bin/chromium-browser](https://github.com/volumio/volumio-plugins/blob/master/plugins/miscellanea/touch_display/install.sh#L75)
However, for TouchDisplay plugin support @volumio is definitely the right person.

biva, 2019-03-30:
@volumio would you have any idea? I can't install Chromium on my Volumio :(
Thanks for your support!