volumio/Volumio2 #1598: autostart plugin?

houndofsound, 2018-08-25:
an autostart plugin has been built in to volumio now.
after reporting that it didn't actually work in v2.444 (x86) 
I now find that it as disappeared from the installed plugins list!
moveover it is no longer listed under 'miscelleanea'
the only configuration changes I have made apart from this are networking
so it is baffling how something could be there one moment, gone the next.

houndofsound, 2018-08-25:
...and now all plugins have disappeared and the search function doesn't offer an input box.
volumio works in mysterious ways....

houndofsound, 2018-08-25:
I think the disappearing plugins were due to the wireless network not being started which in itself is mysterious because it was setup and running ... even between reboots!

and perhaps the autostart plugin is not in the x86 distribution to begin with? (I configured an rpi around the same time so maybe having a false memory of setting autostart up for the x86 also :(

ashthespy, 2018-08-28:
>and perhaps the autostart plugin is not in the x86 distribution to begin with? (I configured an rpi around the same time so maybe having a false memory of setting autostart up for the x86 also :(

Nope, it's not listed for x86 ([`plugins.json`](https://github.com/volumio/volumio-plugins/blob/gh-pages/plugins/volumio/i386/plugins.json)).
Strange though, as it seems to be a pure node plugin, that shouldn't have any cpu architecture specific binaries/dependencies! 

PS: You can always install the plugin manually!

houndofsound, 2018-08-28:
I tried with the github code pointed to on the plugin web page - the
command-line procedure hangs.
perhaps this is different in some way to the code bundled with 2.444
on the RPi ?

volumio, 2018-09-16:
Please contact plugin author

closed 2018-09-16 by volumio