volumio/Volumio2 #1599: Samba misconfiguration ....

houndofsound, 2018-08-25:
the /etc/samba/smb.conf entry for 'Internal Storage' specifies /data/INTERNAL.

it is empty and not a link to anywhere useful.
isn't this wrong considering /etc/mpd.conf says the music directory is /var/lib/mpd/music
under which is a link, INTERNAL, to /mnt/INTERNAL ?
the samba entry should be one or the other of these I think.

xipmix, 2018-09-04:
Not sure all the symlinks are really needed but it does seem at least consistent.
```
$ ls -l /mnt/INTERNAL
lrwxrwxrwx 1 root root 14 Jun 20 23:30 /mnt/INTERNAL -> /data/INTERNAL
$ realpath /var/lib/mpd/music/INTERNAL
/data/INTERNAL
```
The  [docs](https://volumio.github.io/docs/FAQs/Audio_Sources.html) say that area is intended to be used for adding song files local to the SD card. Presumably the intent of the smb.conf entry is to make them shareable on the local network as well.

I'm struggling to see an issue here...

xipmix, 2018-09-05:
See also https://github.com/volumio/Volumio2/issues/939#issuecomment-270016225

houndofsound, 2018-09-07:
There was no data partion mounted under /mnt on the x86 system I installed.
When I did mount it there was no INTERNAL link or subfolder hence why
I thought it odd to see such an entry in smb.conf.

closed 2018-09-16 by volumio