volumio/Volumio2 #1600: Period in artist name/directory name isn't included

maxengel, 2018-08-30:
For example, if I have a directory labeled ```R.E.M.```, the contents aren't being evaluated for inclusion in the library. The share is mounted via SMB. It has showed up in previous versions of Volumio, but seems to not be working in the latest update.

xipmix, 2018-09-01:
Can you please specify what version, and if you know it, a version that was working? That will hugely narrow down the search space.

xipmix, 2018-09-01:
I tested with an NFS mount on 2.413 (see below).

It works ok on artist names that have '.' in them - they show up in the browse->musiclibrary->nas listing and in the artists listing. It does require a 'rescan' and a refresh of the browse page - e.g. navigate to playback and then back to browse.

```
$ cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
VOLUMIO_BUILD_VERSION="d0d14b37ffe8e45861e4e0b9816241fbafb4eb91"
VOLUMIO_FE_VERSION="81c7100e9c7342b16fc9e0ca5908a26771e08c4f"
VOLUMIO_BE_VERSION="645f0e55945bdd625a662a1ee21ba0dfb11bd73e"
VOLUMIO_ARCH="arm"
VOLUMIO_VARIANT="volumio"
VOLUMIO_TEST="FALSE"
VOLUMIO_BUILD_DATE="Thu Jun 21 01:30:10 CEST 2018"

VOLUMIO_VERSION="2.413"
VOLUMIO_HARDWARE="pi"
```

xipmix, 2018-09-01:
After much palaver (we really need to make smb mounting easier) I was able to mount a test smb share, and it too saw the artist folder named j._S.Bach. So the change is probably between 2.413 (june) and 2.449 (august). I'll try to reprocue on 2.449. when I have a bit more time.

xipmix, 2018-09-01:
This is a guess but possibly changing the metadata parsing library has produced this issue?

https://github.com/volumio/Volumio2/commit/6005e90eb181a7d79c8ff16a2eb096e4606184c4

maxengel, 2018-09-03:
Interesting wrinkle:
```volumio@volumio:/mnt/NAS/Media/Acoustic Sounds Super HiRez/R.E.M.$ ls```

xipmix, 2018-09-04:
You can probably address this by disabling 'name mangling' (see links below) but it will be interesting to see if the NFS mount behaves the same, please let us know.
 * https://www.oreilly.com/openbook/samba/book/ch05_04.html
 * https://serverfault.com/questions/124611/special-characters-in-samba-filenames

maxengel, 2018-09-04:
NFS performed much better (no mangling, etc.). I'm going to stick with NFS for now since I'm only in a Linux/Mac world anyways.

Also, the mangling only manifested when browsing by directory and not once the metadata had been fetched and the library built. Also, due to the issue with SMB v3+, I'll wait to test fixing mangling until that issue is resolved.