volumio/Volumio2 #1602: [feature request] Make error message available from UI when mounts fail

xipmix, 2018-09-01:
I was trying to mount a SMB share and it was failing becuase of a missing ```vers=1.0``` mount option. An error is displayed as a toast popup, but far too briefly.

I think it would help the UI if when a mount fails you were able to click on the 'X' icon and view the error message in a modal popup - ie one that persists until you are ready to click 'done'.

xipmix, 2018-09-01:
It might also be nice to display some hints based on the mount typ - for SMB typical problems are vers=1.0 option, wrong username (no username), etc; for NFS it's often incorrect path or wrong case in the path.

xipmix, 2018-09-02:
Example issue where this could help: #1517

Raincode, 2023-08-15:
I second this. Wrong credentials were used.
Only debugging directly on the Pi with 'sudo journalctl -f' finally showed e.g. error code -13, which gave the hint.
Would be awesome if we can somehow take the command output from mount and properly handle different errors as easily viewable message.

volumio, 2023-08-21:
I agree, problem is we don't get any error from the mount command standard output, it's a kernel message, which makes it really complex to catch and parse...