volumio/Volumio2 #1603: Command line api playlist problem

OldDuffer, 2018-09-07:
I have success using the following;

curl volumio.local:3000/api/v1/commands/?cmd=play

also with stop, pause, clearQueue etc. Volumio carries out these commands.

When I try;

curl volumio.local:3000/api/v1/commands/?cmd=playplaylist&name=Playlist1

I get the report "playplaylist success". However no playlist is loaded.

Also if I try to use 'listplaylists' I get the error 'command not recognised'

xipmix, 2018-09-08:
With no playlists defined at all, and testing like so (version 2.452, rpi).
```
$ curl 'http://192.168.x.y:3000/api/v1/commands/?cmd=playplaylist&name=Playlist1'
{"time":1536398623023,"response":"playplaylist Success"}
```
The log contains only this
```
Sep 08 09:23:43 volumio volumio[855]: ---------------------------- Client requests Volumio Play Playlist Playlist1
Sep 08 09:23:43 volumio volumio[855]: info: Play playlist Playlist1
```

If I define a playlist named Playlist1 (select an album from the NAS, use 'add to playlist') and try again I get the same response
```
{"time":1536399134568,"response":"playplaylist Success"}
```
but the playlist gets added to the queue and starts playing. Log shows lots of activity
```
Sep 08 09:28:58 volumio volumio[855]: info:
Sep 08 09:28:58 volumio volumio[855]: ---------------------------- Client requests Volumio Play Playlist Playlist1
Sep 08 09:28:58 volumio volumio[855]: info: Play playlist Playlist1
Sep 08 09:28:58 volumio volumio[855]: info: CoreCommandRouter::volumioClearQueue
Sep 08 09:28:58 volumio volumio[855]: info: CoreStateMachine::ClearQueue
Sep 08 09:28:58 volumio volumio[855]: info: CoreStateMachine::stop
Sep 08 09:28:58 volumio volumio[855]: info: CoreStateMachine::setConsumeUpdateService undefined
Sep 08 09:28:58 volumio volumio[855]: UNSET VOLATILE
Sep 08 09:28:58 volumio volumio[855]: info: CoreStateMachine::stPlaybackTimer
Sep 08 09:28:58 volumio volumio[855]: info: CoreStateMachine::updateTrackBlock
Sep 08 09:28:58 volumio volumio[855]: info: CorePlayQueue::getTrackBlock
Sep 08 09:28:58 volumio volumio[855]: info: CoreStateMachine::pushState
Sep 08 09:28:58 volumio volumio[855]: info: CoreStateMachine::getState
Sep 08 09:28:58 volumio volumio[855]: info: CorePlayQueue::getTrack 0
Sep 08 09:28:58 volumio volumio[855]: info: CoreCommandRouter::volumioPushState
Sep 08 09:28:58 volumio volumio[855]: info: CoreCommandRouter::executeOnPlugin: volumiodiscovery , saveDeviceInfo
Sep 08 09:28:58 volumio volumio[855]: info: CoreStateMachine::serviceStop
Sep 08 09:28:58 volumio volumio[855]: info: CorePlayQueue::getTrack 0
Sep 08 09:28:58 volumio volumio[855]: info: CoreCommandRouter::serviceStop
Sep 08 09:28:58 volumio volumio[855]: info: ControllerMpd::stop
Sep 08 09:28:58 volumio volumio[855]: info: ControllerMpd::sendMpdCommand stop
Sep 08 09:28:58 volumio volumio[855]: info: CorePlayQueue::clearPlayQueue
Sep 08 09:28:58 volumio volumio[855]: info: CorePlayQueue::saveQueue
Sep 08 09:28:58 volumio volumio[855]: info: CoreStateMachine::pushEmptyState
Sep 08 09:28:58 volumio volumio[855]: info: CoreCommandRouter::volumioPushState
Sep 08 09:28:58 volumio volumio[855]: info: CoreCommandRouter::executeOnPlugin: volumiodiscovery , saveDeviceInfo
Sep 08 09:28:58 volumio volumio[855]: info: CoreCommandRouter::volumioPushQueue
Sep 08 09:28:58 volumio volumio[855]: info: CoreCommandRouter::volumioAddQueueItems
Sep 08 09:28:58 volumio volumio[855]: info: CoreStateMachine::addQueueItems
Sep 08 09:28:58 volumio volumio[855]: info: CorePlayQueue::addQueueItems
Sep 08 09:28:58 volumio volumio[855]: info: Adding Item to queue: mnt/NAS/welly/U2/Rattle_and_Hum/01-Helter_Skelter.flac
Sep 08 09:28:58 volumio volumio[855]: info: Exploding uri mnt/NAS/welly/U2/Rattle_and_Hum/01-Helter_Skelter.flac in service mpd
Sep 08 09:28:58 volumio volumio[855]: info: Adding Item to queue: mnt/NAS/welly/U2/Rattle_and_Hum/02-Van_Diemen_s_Land.flac
Sep 08 09:28:58 volumio volumio[855]: info: Exploding uri mnt/NAS/welly/U2/Rattle_and_Hum/02-Van_Diemen_s_Land.flac in service mpd
Sep 08 09:28:58 volumio volumio[855]: info: Adding Item to queue: mnt/NAS/welly/U2/Rattle_and_Hum/03-Desire.flac
Sep 08 09:28:58 volumio volumio[855]: info: Exploding uri mnt/NAS/welly/U2/Rattle_and_Hum/03-Desire.flac in service mpd
Sep 08 09:28:58 volumio volumio[855]: info: Adding Item to queue: mnt/NAS/welly/U2/Rattle_and_Hum/04-Hawkmoon_269.flac
Sep 08 09:28:58 volumio volumio[855]: info: Exploding uri mnt/NAS/welly/U2/Rattle_and_Hum/04-Hawkmoon_269.flac in service mpd
Sep 08 09:28:58 volumio volumio[855]: info: Adding Item to queue: mnt/NAS/welly/U2/Rattle_and_Hum/05-All_Along_the_Watchtower.flac
Sep 08 09:28:58 volumio volumio[855]: info: Exploding uri mnt/NAS/welly/U2/Rattle_and_Hum/05-All_Along_the_Watchtower.flac in service mpd
Sep 08 09:28:58 volumio volumio[855]: info: Adding Item to queue: mnt/NAS/welly/U2/Rattle_and_Hum/06-I_Still_Haven_t_Found_What_I_m_Looking_For.flac
...
Sep 08 09:28:58 volumio volumio[855]: info: sending command...
Sep 08 09:28:58 volumio volumio[855]: info: parsing response...
Sep 08 09:28:58 volumio volumio[855]: info:
Sep 08 09:28:58 volumio volumio[855]: ---------------------------- MPD announces state update: player
Sep 08 09:28:58 volumio volumio[855]: info: ALBUMART /albumart?cacheid=138&web=U2/Rattle%20and%20Hum/extralarge&path=%2Fmnt%2FNAS%2Fwelly%2FU2%2FRattle_and_Hum%2F01-Helter_Skelter.flac&metadata=false
Sep 08 09:28:58 volumio volumio[855]: info: URI /mnt/NAS/welly/U2/Rattle_and_Hum/01-Helter_Skelter.flac
Sep 08 09:28:58 volumio volumio[855]: info: ALBUMART /albumart?cacheid=138&web=U2/Rattle%20and%20Hum/extralarge&path=%2Fmnt%2FNAS%2Fwelly%2FU2%2FRattle_and_Hum%2F02-Van_Diemen_s_Land.flac&metadata=false
Sep 08 09:28:58 volumio volumio[855]: info: URI /mnt/NAS/welly/U2/Rattle_and_Hum/02-Van_Diemen_s_Land.flac
[... cut ...]
Sep 08 09:28:58 volumio volumio[855]: info: CoreStateMachine::pushState
Sep 08 09:28:58 volumio volumio[855]: info: CoreStateMachine::getState
Sep 08 09:28:58 volumio volumio[855]: info: CorePlayQueue::getTrack 0
Sep 08 09:28:58 volumio volumio[855]: info: CoreCommandRouter::volumioPushState
Sep 08 09:28:58 volumio volumio[855]: info: CoreCommandRouter::executeOnPlugin: volumiodiscovery , saveDeviceInfo
Sep 08 09:28:58 volumio volumio[855]: info: ------------------------------ 223ms
```

Can you send logs of what happens on your system?

OldDuffer, 2018-09-08:
Many thanks, I'll looking into the log sending process and see what I can do!
Am I right in thinking you did not have an actual playlist 'Playlist1' for the first attempt?

OldDuffer, 2018-09-08:
Sorry! Re read and realised that was a silly question.  Will send logs as soon as I found out how.

OldDuffer, 2018-09-08:
I should also have put in the query that the playlist 'Playlist1' loads and plays when the command is issued from a web browser.

xipmix, 2018-09-09:
Sending logs - https://volumio.github.io/docs/User_Manual/Sending_logs_for_troubleshooting.html
You can also ssh in and run 'volumio logdump' fthe same report sent to a local file.

xipmix, 2018-09-09:
Hm. I had a browser open and pointed at the volumio instance while I was running the curl tests. Maybe I will try with no browser open, just curl.

xipmix, 2018-09-10:
I gave the curl-only thing a try, it worked ok.

chsims1, 2018-09-11:
Works if you include the single quotes, but doesn't without (even though responds with "success").

OldDuffer, 2018-09-11:
Thank you for your time and effort.

closed 2018-09-11 by OldDuffer