volumio/Volumio2 #1604: odd results when using setSleep via socket.io API

sstark, 2018-09-10:
If you set a sleep timer using the websocket API like this:

`emit('setSleep', {"enabled":"true", "time":"00:02", "action":"stop"})`

volumio will correctly use that sleep timer. However, the output of getSleep looks wrong:

`
[
    {
        "action": {
            "text": "Stop Music",
            "val": "stop"
        },
        "enabled": "true",
        "time": "0:0"
    }
]
`

As you can see, the value for _time_ is still set to 0:0, but it should read 0:1 or 0:2 immediately after setting the sleep timer. Also, if you use the webgui to check the sleep timer, it is shown as not set and inactive.

If using the web gui to set the sleep timer, the output of getSleep is correct.

I am using the x86 version from https://hub.docker.com/r/jbonjean/volumio/ (updated 25 days ago) and python3-socketio-client to test this.

whxciotw, 2019-09-28:
Hi, I have something confused about the websocket URL, and I want to know what's your URL?
is it ws://localhost:3000? or something else?

It would be appreciated for your reply, thanks.

sstark, 2019-09-30:
Sorry, I don't use volumio anymore as it had too many bugs for me and too little response to my reports.