volumio/Volumio2 #1608: No Sound with HiFiBerry Amp+ and Volumio2 2.452 Upgrade

billdenney, 2018-09-16:
After upgrading from Volumio2 around version 2.387 (I'm not sure what the specific prior version was) to 2.452, I lost sound.  

As described in https://volumio.org/forum/sound-with-volumio-452-upgrade-and-hifiberry-amp-t10705.html, I can get audio from command line programs, so it does not appear to be a hardware or non-volumio software issue.

The issue appears likely due to something in the playback options:
[image]

I've tried changing the options and saving them in case they somehow got into a corrupt state and could be changed back, but after saving, they go back to what is in the screenshot.

billdenney, 2018-09-21:
After doing a factory reset and replacing all my settings, it worked again.

Is there a way to prevent such invalid settings from being set on upgrade in the future?

xipmix, 2018-09-23:
Glad  you got it going. 
Do you know  which particular settings were changed to the 'wrong thing' by the upgrade? That would help determining how to fix this. 
Alternatively, it may be helpful to list what settings you had to change from the defaults to get working sound.
Also I am curious  what this command gives:
```
$ ssh volumio@volumio
$ cd /volumio; git log -1; git diff; git status
```

billdenney, 2018-09-23:
Unfortunately, I have no idea which settings were changed by the upgrade.  My only thought was that the above screen shot may help since I couldn't modify any of those settings.  I had previously set the Audio Buffer Size (I don't recall the value-- but I think it was the largest available) and Buffer Before Play (again, I think it was the largest value available).  Those were a work-around for a previous unreliable wifi network.

After the factory reset, the settings that I changed were:
* Playback Options
    * I2S DAC: On
    * DAC Model: HiFiBerry Amp
    * Save
(reboot)

After that, it worked.  (I have made some other changes around startup volume and audio buffering, but audio was working before making those changes, so I assume that they aren't relevant.  If they are relevant, let me know, and I will enumerate them.)

```
volumio@livingroom:~$ cd /volumio/
volumio@livingroom:/volumio$ git log -1
commit 909cc76a7609428f005f8e2878227062cb5328a6
Author: Volumio <info@volumio.org>
Date:   Sun Sep 16 21:15:26 2018 +0000

    Adding cymatic bg
volumio@livingroom:/volumio$ git diff
volumio@livingroom:/volumio$ git status
On branch master
Your branch is up-to-date with 'origin/master'.
Untracked files:
  (use "git add <file>..." to include in what will be committed)

        app/plugins/system_controller/network/wireless.js
        http/www/

nothing added to commit but untracked files present (use "git add" to track)
volumio@livingroom:/volumio$
```

xipmix, 2018-09-24:
Thanks for that, I was wondering if something had changed settings buried in code someplace but that seems not to be the case - which is a good thing. 

A little poking around shows the settings likely live in /data/configuration/audio_interface/alsa_controller/config.json and/or /data/configuration/system_controller/i2s_dacs/config.json (and maybe also /data/configuration/plugins.json). So perhaps upgrading somehow touches those files. Looking at the code in the git repo I do see app/plugins/system_controller/i2s_dacs/config.json which has some default settings but nothing that looks like it would override the settings stored in /data.

It's odd that you could not fix things by adjusting settings in the web ui - that should update the files I mentioned and those files are how the settings are persisted across reboots. I assume you tried a simple reboot before the factory reset, and that had no effect?

xipmix, 2018-09-25:
I'd guess the volumio log, captured after you reboot & try to make changes in the uncooperative web UI. ( [instructions](https://volumio.github.io/docs/User_Manual/Sending_logs_for_troubleshooting.html) ). Plus all the files in /data; ```tar cf data.tar /data```. Not sure how the data could be shared with the devs, maybe dropbox or something.

Maybe @volumio has some suggestions too? It sounds like something was up with the Volumio2-UI code across that transition (that would cause the UI strangeness) but I have no idea what to look for.

jan-coppens, 2020-04-09:
Hi, 

I experience the same issue as described above.
I use a HifiBerry Amp+ on a RaspberryPi Model 3B. (with the latest firmware 2.729)

I already tried to do a factory reset but the issue remains.