volumio/Volumio2 #1612: Consistant code style

ashthespy, 2018-09-19:
As Volumio grows, so does it's codebase. 
Going by this nice quote from [`idiomatic.js`](https://github.com/rwaldron/idiomatic.js/):
>All code in any code-base should look like a single person typed it, no matter how many people contributed.

Would there be a Volumio specific style guide? If not, [`standardjs`](https://github.com/standard/standard) seems to be the most widely used, and has wide IDE support!

volumio, 2018-09-19:
Good idea. Would you like to help setting it up and fixing code?

volumio, 2018-09-19:
I feel silly, but... https://github.com/Flet/semistandard ?

Hate to say goodbye to semicolons...

ashthespy, 2018-09-19:
Haha, I just linted some code and it felt strange not to have semicolons. 

In terms of workflow, we could add a CI to run and commit the lints, or use some git hooks to run `standard --fix` before any commits. But then in depends on the devs individual workflow.

volumio, 2018-09-19:
Deal! Semistandard and githooks!

closed 2018-09-20 by ashthespy