volumio/Volumio2 #1616: System Software Integrity check failed

suuuehgi, 2018-09-25:
> Update failed
> System Software Integrity check failed, manually edited files were detected. System Software Update is not possible under those circumstances. Please execute a Factory Reset or Delete User Data, then retry.

The message is too unspecific for me to result in action. Neither will I delete all my settings, nor did I change something besides the `/boot/config.txt` and the favourites in `/data/favourites/`. Both shouldn't be a problem in the update process.

Is there a way to get more information about the issue? This is the first time while updating this message appears.

System version: 2.444

volumio, 2018-09-25:
We had other reports of a potential issue with the update integrity check.

I prepared a script that will list which files were edited.
Please do this:

- Enable SSH ( https://volumio.github.io/docs/User_Manual/SSH.html )
- Connect via SSH to Volumio and type those 2 commands:

```
cd /volumio
wget -O - http://repo.volumio.org/Volumio2/Test-tools/listdynchanges.sh | bash
```

It's very useful for us to know which files were modified. Please post the result of the above command here.

After that, you can fix your situation by issuing the following command via SSH:
`volumio updater restore volumio`

suuuehgi, 2018-09-25:
Thanks a lot for the effort!

The problem file is `/mnt/overlay/dyn/volumio/app/statemachine.js`.

volumio, 2018-09-25:
This means you edited it manually ;)

closed 2018-09-25 by volumio

suuuehgi, 2018-09-25:
> This means you edited it manually ;)

Definitely not!

suuuehgi, 2018-09-27:
Hm ... after the update it rebootet and became unresponsive. That means nothing on the attached screen and no reaction on ssh. No change after several reboots. Then I let it run and now, half a day later when I came home, it works again fine. Do you have any idea about this?

Anyway, it works again now! :-) Thank you very much!