volumio/Volumio2 #1627: Change plugin `install.sh` shell to bash?

ashthespy, 2018-10-11:
Any reason for not using `bash` but `sh` to execute a plugins's `install.sh`?
https://github.com/volumio/Volumio2/blob/1e43e1e14289e4237ffdc30fc2579d650d530e64/app/pluginmanager.js#L1310-L1311

A quick grep shows that most of the plugin install scripts want to use bash:
```shell
1:#!/bin/bash
plugins/audio_interface/volsimpleequal/uninstall.sh
plugins/audio_interface/volsimpleequalmbeq/uninstall.sh
plugins/audio_interface/volsimpleequal/install.sh
plugins/audio_interface/volsimpleequalmbeq/install.sh
plugins/system_controller/volsmartpower/install.sh
plugins/system_controller/volsmartpower/uninstall.sh
plugins/audio_interface/volumiobauerfilter/uninstall.sh
plugins/miscellanea/volumiominidlna/install.sh
plugins/miscellanea/volumiominidlna/uninstall.sh
plugins/audio_interface/volumiobauerfilter/install.sh
plugins/music_service/volspotconnect/install.sh
plugins/music_service/volspotconnect/umountchroot.sh
plugins/music_service/volspotconnect/uninstall.sh
plugins/music_service/volspotconnect2/install.sh
plugins/music_service/volspotconnect2/onstart1.sh
plugins/music_service/volspotconnect2/uninstall.sh
plugins/audio_interface/volparametriceq/install.sh
plugins/audio_interface/volparametriceq/uninstall.sh
plugins/audio_interface/brutefir2/uninstall.sh
plugins/audio_interface/brutefir2/install.sh
plugins/music_service/spotify/install.sh
plugins/music_service/spotify/uninstall.sh
plugins/system_controller/wifireconnect/install.sh
plugins/system_controller/wifireconnect/uninstall.sh
plugins/audio_interface/volgrp/install.sh
plugins/audio_interface/volgrp/uninstall.sh
plugins/audio_interface/volstereo2mono/install.sh
plugins/audio_interface/volstereo2mono/uninstall.sh
plugins/miscellanea/playonconnect/install.sh
plugins/miscellanea/playonconnect/uninstall.sh

```

volumio, 2018-10-12:
I don't remember any particular reason, feel free to experiment with bash!

xipmix, 2018-10-13:
I have my doubts about this because most invocations of such scripts are done by calling ```/bin/sh /path/to/script``` which means the ```#!``` line will be ignored. There are a lot of calls like this within volumio. Those scripts, if they are working when invoked with /bin/sh already follow the standard Bourne shell syntax and if they used any bash-only features they would fail. 

So I wonder what the benefit of using bash is - can you explain? A more sustainable course of action might be to just ensure the scripts all have the execute bit set and remove the ```/bin/sh``` prefixes when calling them. Then they will run under whatever interpreter the ```#!``` line defines, which could even be python or perl, or ... node??

ashthespy, 2018-10-13:
@xipmix exactly my point - these scripts all ask for bash with the shebang but is ignored. For example I was using associative arrays in my install script and was wondering why it wasn't working when called by the plugin installer. 

Your proposal would be `chmod +x installScript && echo volumio | sudo installScript` ?

xipmix, 2018-10-13:
I was thinking more that plugins would be required to set the execute bit but what you suggest would work too. This seems like a good time to try  to get rid of the ```echo volumio```, as well. Do you have any ideas about that? see for example discussion in #1247 .

ashthespy, 2018-10-14:
@xipmix I agree getting rid of the `echo volumio` will finally let users change the default login. But I don't have any bright ideas. 
We could just pop a modal on the WebUI. But that would mean the password is sent in plain text of the websocket..

ashthespy, 2018-10-17:
@xipmix Should I open up a PR with the `echo volumio | sudo` solution, or wait for the fix your proposed?

xipmix, 2018-10-18:
Definitely don't wait for me, go ahead with the PR and we shall see what volumio thinks

ashthespy, 2022-01-22:
Repo (but not old issues) has been moved over to https://github.com/volumio/volumio3-backend

closed 2022-01-22 by ashthespy