volumio/Volumio2 #1629: Can't access web ui after system update

linzack, 2018-10-12:
Hi,
when I open `http://192.168.0.14/playback` or `http://192.168.0.14/dev`
chrome says `ERR_CONNECTION_RESET`, no web ui shows, but `ssh 192.168.0.14` is working.
Everything is ok before I triggered system update this time. 
My last system update is months ago.
I find this in `/var/log/syslog`
```
Oct 13 06:01:24 localhost volumio[5991]: /volumio/node_modules/bindings/bindings.js:83
Oct 13 06:01:24 localhost volumio[5991]: throw e
Oct 13 06:01:24 localhost volumio[5991]: ^
Oct 13 06:01:24 localhost volumio[5991]: Error: The module '/volumio/node_modules/libxmljs/build/Release/xmljs.node'
Oct 13 06:01:24 localhost volumio[5991]: was compiled against a different Node.js version using
Oct 13 06:01:24 localhost volumio[5991]: NODE_MODULE_VERSION 48. This version of Node.js requires
Oct 13 06:01:24 localhost volumio[5991]: NODE_MODULE_VERSION 57. Please try re-compiling or re-installing
Oct 13 06:01:24 localhost volumio[5991]: the module (for instance, using `npm rebuild` or `npm install`).
Oct 13 06:01:24 localhost volumio[5991]: at Object.Module._extensions..node (module.js:681:18)
Oct 13 06:01:24 localhost volumio[5991]: at Module.load (module.js:565:32)
Oct 13 06:01:24 localhost volumio[5991]: at tryModuleLoad (module.js:505:12)
Oct 13 06:01:24 localhost volumio[5991]: at Function.Module._load (module.js:497:3)
Oct 13 06:01:24 localhost volumio[5991]: at Module.require (module.js:596:17)
Oct 13 06:01:24 localhost volumio[5991]: at require (internal/module.js:11:18)
Oct 13 06:01:24 localhost volumio[5991]: at bindings (/volumio/node_modules/bindings/bindings.js:76:44)
Oct 13 06:01:24 localhost volumio[5991]: at Object.<anonymous> (/volumio/node_modules/libxmljs/lib/bindings.js:1:99)
Oct 13 06:01:24 localhost volumio[5991]: at Module._compile (module.js:652:30)
Oct 13 06:01:24 localhost volumio[5991]: at Object.Module._extensions..js (module.js:663:10)
Oct 13 06:01:24 localhost systemd[1]: volumio.service: main process exited, code=exited, status=1/FAILURE
Oct 13 06:01:24 localhost systemd[1]: Unit volumio.service entered failed state.

```

Thank you!

xipmix, 2018-10-13:
It may help to have exact version details. Can you ssh in and send the result of these commands please?
```
$ cd /volumio
$ git status
$ git log -1
$ cat /etc/os-release
```

linzack, 2018-10-13:
ah, yes.
I have tried to `npm install libxmljs` after I saw the error log.
But I dont do git pull.
Is there a solution for this situation?
Thank you!

volumio, 2018-10-13:
yes, volumio updater restorevolumio

linzack, 2018-10-13:
Thank you!
but the output seems like it does not work on my version number
```
volumio@volumio:~$ volumio updater restorevolumio

Usage : volumio <argument1> <argument2>

[[PLAYBACK STATUS]]

status                             Gives Playback status information
volume                             Gives Current Volume Information
volume <desired volume>            Sets Volume at desired level 0-100
volume mute                        Mutes
volume unmute                      Unmutes
volume plus                        Increases Volume of one step
volume minus                       Decreases Volume of one step
seek plus                          Forwards 10 seconds in the song
seek minus                         Backwards 10 seconds in the song
seek <seconds>                     Plays song from selected time
repeat                             Toggles repetition of queue
random                             Toggles randomization of queue

[[PLAYBACK CONTROL]]

play
pause
next
previous
stop
clear

[[VOLUMIO SERVICE CONTROL]]

start                               Starts Volumio Service
vstop                               Stops Volumio Service
restart                             Restarts Volumio Service

[[VOLUMIO DEVELOPMENT]]

pull                               Pulls latest github status on master from https://github.com/volumio/Volumio2.git
pull -b <branch>                   Pulls branch <branch> from https://github.com/volumio/Volumio2.git
pull -b <branch> <repository>      Pulls branch <branch> from git repository <repository>
kernelsource                       Gets Current Kernel source (Raspberry PI only)
plugin init                        Creates a new plugin
plugin refresh                     updates plugin in the system
plugin package                     compresses the plugin
plugin publish                     publishes the plugin on git

```

volumio, 2018-10-13:
Then you definetely messed up the system too much with manual edits.
Only way is to reflash, sorry

linzack, 2018-10-13:
ooops, ok.
Thank you for the help !!

closed 2018-10-13 by volumio

volumio, 2018-10-13:
Yes but I suggest to reflash anyway

linzack, 2018-10-13:
Haha, I get it! Thank you!