volumio/Volumio2 #1639: webradio: playback stops randomly

moritzmhmk, 2018-11-01:
The webradio does stop playback after some time (the UI does NOT reflect that - i.e. the button is still the stop rectangle). Pressing stop and than play will make the stream play again.

~~Using journalctl I found ``volumio wpa_supplicant[778]: wlan0: WPA: Group rekeying completed with xx:xx:xx:xx:xx:xx [GTK=CCMP]`` about a minute later.~~ Probably unrelated - just took 4min for this message to appear... (there are no other messages at all since I have the webui closed).

xipmix, 2018-11-02:
The message you quote indicates some kind of wireless network issue. Do you have music in a local network share connected to your volumio, also via wireless? If that also stops randomly with the message above that suggests it's an issue intrinsic to the wireless. Is it possible to use an ethernet cable to connect your volumio to your home router - if you do that does the problem go away?

moritzmhmk, 2018-11-12:
The same problems occurred when using ethernet.
I found after some debugging that my router has those small drops roughly every 10 minutes. In the end I found that another device caused this - removing the static IP and using DHCP on that device fixed this (for whatever reason).
Volumio while not responsible for the drops does not handle them (at least for the webradio) and will just hang instead of either continuing playback or actually going in a non-playing mode. I fixed this for now by installing [mpd_watchdog](https://github.com/moritzmhmk/mpd-watchdog)

TL; DR
The Internet drops are not caused by volumio but the webradio player does not handle drops. Can be sort of fixed with [mpd_watchdog](https://github.com/moritzmhmk/mpd-watchdog).

elcuevero, 2018-11-13:
@moritzmhmk not interested in hijacking this, but could you post instructions on how to install mpd_watchdog? tried this but got a gpg key error:

wget -q -O - http://apt.kemmer.co.at/tkem.gpg | sudo apt-key add -
sudo wget -q -O /etc/apt/sources.list.d/tkem.list http://apt.kemmer.co.at/tkem.list
sudo apt-get update && sudo apt-get install mpd-watchdog

moritzmhmk, 2018-11-15:
I didn't install it with apt but instead cloned the repo and executed the script.