volumio/Volumio2 #1641: Automounted USB is shown in UI, but appears empty.

almereyda, 2018-11-04:
When attaching a USB drive with an ext4 partition, it is automatically mounted into `/media/` and a USB entry appears in http://volumio/browse.

Unfortunately it is being displayed empty, despite the file system shows all files are owned by user `1000` and have `rwx` permission for that.

xipmix, 2018-11-05:
Good to know about this, thank you for reporting it. Can we see [logs](https://volumio.github.io/docs/User_Manual/Sending_logs_for_troubleshooting.html) please for what happens in this sequence of events:
 * clean reboot
 * insert usb stick
 * visit USB in /browse page

almereyda, 2018-11-06:
I can report back on this by the end of next week.

xipmix, 2018-11-06:
That's fine. If I get a little time I will try to reproduce. Anyone else who can, please comment.

volumio, 2018-11-07:
Known issue, we're working on that in this branch https://github.com/volumio/Volumio2/tree/udevMounter

rayrrr, 2019-05-06:
Any news on this? The branch link results in a 404 page...

laurentyhuel, 2019-07-18:
I meet same problem, with an USB external HDD in Ext4 (1To) and a small USB key in Ext4. Is branch udevMounter is merged and closed ?

WarmChocolateCake, 2019-10-02:
> Known issue, we're working on that in this branch https://github.com/volumio/Volumio2/tree/udevMounter

What is the status of this now? The URL returns a 404 and v2.619 appears to have this problem

WarmChocolateCake, 2019-10-03:
Just adding that volumio is "seeing" the USB media because I can stream it on another device (android tablet running VLC)
So, (in my case at least) this appears to be an issue with the UI not udev??

rayrrr, 2019-10-03:
Stupid question, but it hasn't been asked in this thread yet so I'll go ahead...does scanning the music library after plugging in the USB storage device make any difference?

volumio, 2019-10-03:
You're having EXT4, so probably you use those on linux.
Can you check the permissions on the files? mpd requires to access those files

WarmChocolateCake, 2019-10-03:
> does triggering a rescan of the music library after plugging in the USB storage device make any difference?

Nope, but good question... when I triggered a Rescan, watching `nmon` showed no activity on the USB drive at all... so volumio isn't even trying to access the drive.

> Can you check the permissions on the files? mpd requires to access those files

Aha! Checking `systemctl list-units` shows that `mpd` had failed... (sorry, was absorbed by device mounts, didn't think to check services...)

Long story short, it looks like `mpd` was broken trying to access `libbcm_host.so` as per this post: [https://forum.volumio.org/mpd-broken-volumio-t11545.html](url)

Re-linking the library appears to have solved this issue (for me)

Thanks for the pointer @volumio  :)

askdesigners, 2020-04-11:
I'm seeing this also now in april 2020. It looks like maybe it's a slow scan with no  status indication? I had it running all day yesterday though and it only found the first couple of folders. Looking at the drive, there are quite some hidden files (osx trash, etc) so I was thinking if I renamed folders with a leading _ to get the to appear before the hidden files maybe that would work, and it seems to! 

All files that fall after the `.filename` hidden files never show up. My guess is that it's crawling the dir and dies when it hits hidden files...