volumio/Volumio2 #1644: DSD and FLAC in one playlist

novtom, 2018-11-05:
Hi, 
If I setup playlist from different file types, I am not able to play this order:

file1.dsf
file2.flac (16bit 44,1kHz)

From 16 bit to dsf or to 24 bit FLAC it is ok, also from 24 bit FLAC to 16 bit FLAC is everything all right. This combinations are working:
file1.dsf
file2.flac (24 bit - all kHz (96, 48, 176, 192)) 

file1.dsf
file2.dsf

file1.flac (16 bit)
file2.flac (24 bit)

file2.flac (24 bit)
file1.flac (16 bit)

This update I have also tried:

```
ssh root login

cd /usr/src

wget https://raw.githubusercontent.com/sparky-sbc/sparky-test/master/pro-ject-s2/snd-usb-audio.ko -O /lib/modules/3.10.38/kernel/sound/usb/snd-usb-audio.ko

wget https://raw.githubusercontent.com/sparky-sbc/sparky-test/master/pro-ject-s2/snd-usbmidi-lib.ko -O /lib/modules/3.10.38/kernel/sound/usb/snd-usbmidi-lib.ko

sync

reboot
```

But switching between DSD and 16 bit doesn't work. It shows me error opening ALSA device snd_pcm_hw_params () failed.
Thanks a lot

Tomas

xipmix, 2018-11-06:
Thanks for your detailed report. It would be good to see the exact error messages and their context, can you please send some example [logs](https://volumio.github.io/docs/User_Manual/Sending_logs_for_troubleshooting.html)?

novtom, 2018-11-06:
Hello, I have made a log. Link is here:
[log](http://logs.volumio.org/volumio/v2enfjJ.html)

novtom, 2019-01-03:
Please close this topic. Now it has been repaired. The problem wasn't in Volumio, but in old firmware in Pro-Ject MaiA DS2. I have upgraded to new 1.07 firmware and it is OK.