volumio/Volumio2 #1645: Musicbrainz tag typo

antlarr, 2018-11-08:
There seems to be a typo at:
https://github.com/volumio/Volumio2/blob/99d5a51bbd1ace8ca0d309425ef1210697a04639/app/plugins/user_interface/mpdemulation/helper.js#L121

The correct name seems to be defined at:
https://github.com/MusicPlayerDaemon/MPD/blob/e28d1e0f65fb47f452934bc5d70fdb40bdc4c62e/src/tag/Names.c#L43

Also, there seems to be a few more musicbrainz tags used by mpd (MUSICBRAINZ_ALBUMID, MUSICBRAINZ_RELEASETRACKID, MUSICBRAINZ_WORKID). It would be great if Volumio used them.

xipmix, 2018-11-10:
Question - how did you notice this - was it just perusing the code or was some functionality not working for you?

closed 2018-11-18 by volumio

antlarr, 2018-11-19:
> Question - how did you notice this - was it just perusing the code or was some functionality not working for you?

Hehe, actually, I'm afraid I have never used Volumio. I just knew about it and wanted to know what level of MusicBrainz integration it has, as I've been thinking on developing a music player with MusicBrainz integration (or helping develop one if I find any advanced player that has some support already). So I searched into the code for "MusicBrainz", noticed this looked like a  typo, checked in the mpd code if it's actually a typo and I thought you (the developers) would like to know about it so I reported it.

commit 6526a357 (2018-11-18): Musicbrainz (#1647)

* Typo. Fixes: #1645

* Additional musicbrainz tags requested in #1645

* Comment out MUSICBRAINZ_WORKID

The version currently in volumio is 0.20.6