volumio/Volumio2 #1646: Multiroom Devices not reachable with Volumio 2.502

etuo, 2018-11-09:
Multiroom Devices not reachable with Volumio 2.502
They will be shown but if You choose one it´s loading endless without success.

etuo, 2018-11-10:
http://logs.volumio.org/volumio/rh7V5VN.html

thanks

发自我的 iPhone

xipmix, 2018-11-10:
I don't know anything about how the multiroom plugin works, so I won't be much more help but I did do a quick scan of the logs. The only thing that seemed out of order was this addMembership error near the start.
```
Nov 09 17:40:41 volumio volumio[888]: info: CoreCommandRouter::executeOnPlugin: multiroom , getMultiroom
Nov 09 17:40:41 volumio volumio[888]: info: Listing playlists
Nov 09 17:40:41 volumio volumio[888]: info: CoreCommandRouter::volumioGetQueue
Nov 09 17:40:41 volumio volumio[888]: info: CoreStateMachine::getQueue
Nov 09 17:40:41 volumio volumio[888]: info: CorePlayQueue::getQueue
Nov 09 17:40:41 volumio shairport-sync[1037]: Successful Startup
Nov 09 17:40:41 volumio volumio[888]: info: Shairport-Sync Started
Nov 09 17:40:41 volumio volumio[888]: Error adding Membership: Error: addMembership EINVAL
Nov 09 17:40:41 volumio volumio[888]: info: Shairport-Sync Started
Nov 09 17:40:41 volumio volumio[888]: Discovery: adding 82ce2717-43d0-44d6-9aea-1cf2e608dec4
Nov 09 17:40:41 volumio volumio[888]: info: mDNS: Found device Volumio
```
I was also wondering how your multiroom devices are connected together and how many there are
 * is your volumio using wifi, ethernet or both?
 * are the other devices all using the same medium (wifi or ethernet)
 * how many devices, including volumio are involved? Do any play sound (it seems mot from your report)
Sorry for all the questions,  just trying to make the search space smaller.

etuo, 2018-11-10:
I have 3 pi, a pi 3b+， a pi zero, a pi 3b, all using wifi networking
, the IP address is automatically obtained through dhcp
There are more than 10 devices connected to the router including volumio
Multiroom Other volumio IP address is wrong, it is a non-existent ip

ashthespy, 2018-11-20:
Also noticed this - I can see the `2.502` device from my other devices, but once I navigate to it, it can't see the other devices.
Logs show:
```
info: CoreCommandRouter::executeOnPlugin: multiroom , getMultiroom
Plugin multiroom or method getMultiroom not found
```

ashthespy, 2018-11-21:
@xipmix I was digging around last night as I didn't see any `multiroom` plugin under `audio_interface` - unless I am missing something, this would have always failed?

If that is indeed the case, the new entries in the log show up only because the new version (myvolumio) logs an error:

https://github.com/volumio/Volumio2/blob/b92c3a7d5bde601d9be147adb62fbab3d14868aa/app/plugins/user_interface/websocket/index.js#L910-L922

while the old one doesn't:

https://github.com/volumio/Volumio2/blob/74c3e92284178bbb6bc37c07d3d4faafc7678639/app/plugins/user_interface/websocket/index.js#L742-L754

ashthespy, 2018-11-21:
>@xipmix I was digging around last night as I didn't see any multiroom plugin under audio_interface - unless I am missing something, this would have always failed?
>
>If that is indeed the case, the new entries in the log show up only because the new version (myvolumio) logs an error:

FWIW `returnedData` is always `undefined`, even on the working build.