volumio/Volumio2 #1648: myvolumio favorites sync problem

CrYpT0n1T3, 2018-11-10:
my volumio restores my favorites even though I removed them
it's very annoying, I remove it and myvolumio restores it all over again

(rpi 3, volumio 2.502 )

volumio, 2018-11-13:
Thanks for notifying. Can you please write the exact steps to reproduce the issue?

CrYpT0n1T3, 2018-11-13:
Example.
1. I create a list called OLDSONGS and add a few songs to it.
2. I want to change the name of the list but I can not, so I create a different one and remove OLDSONGS.
3. OLDSONGS is no longer on the list.
4. I listen to music, browse collections.
5. I look into the list after a few minutes and the OLDSONGS list appears again.
6. I remove it.
7. After a few minutes he appears again.
8. Over and over
The problem only applies when I am logged in to myvolumio.

Synchornization must mean which lists have been removed and not restored after re-synchronization.