volumio/Volumio2 #1649: volumio does not show album covers

CrYpT0n1T3, 2018-11-14:
I have two pendrives connected.
Total 362 albums. When I enter a collection of music, Volumio does not show all the covers. Each album has a file called cover.jpg or cover.png smaller than 1mb, mainly 300x300px.
Volumio does not show me random album covers, mainly those at the end. Is there a limit? When I clear the cache album the problem appears on the second pendrive. The problem is changing one on the first memory later on the other.
Memory cleaning in firefox does not help.

Similarly, when I enter albums in the tab. There are no covers for what are in the collection of music. It's like it's random.
And every folder with the album has cover.jpg. I do not understand.
I will be grateful for every hint.

volumio 2.502. rpi3b

CrYpT0n1T3, 2018-11-29:
The problem returned on the 16gb card.
I have a message that there is no free space after apt-get update.
I know where the problem is.
Installation on linux by the ether program - the problem is in the partition created by the installer - volumio_data
Volumio_data is too small and additionally the installer does not allocate a free space partition.
After installation, there is approximately 12 gb of unallocated space on the 16gb card.
The solution is to manually resize the volumio_data partition and assign it the rest of the free space. - gparted

CrYpT0n1T3, 2018-12-03:
etcher-electron-1.4.4-x86_64
,
yes

volumio, 2018-12-03:
Did you install other utilities on the same SD Card? Did you copy music?

CrYpT0n1T3, 2018-12-03:
I did not install anything.
I only have music on a pendrive

If you have ssh to volumio, connect and check how much space you have on the card with the "df" command.
I had only 8mb of free space in "overlay" and I could not even do apt-get update (20mb) and the covers did not show up. I put a card in my laptop in gparted, I noticed that there is 13 gb not allocated to any partition, so I increased volumio_data for this free space and now "overlay" size 13gb and use 3%. So a problem is with the installation because the problem has repeated on 8gb and 16gb.

volumio, 2018-12-03:
Last partition gets resized to fill whole sd card space on first boot...

volumio, 2018-12-03:
Installer? Are you using an official volumio image or another kind of installer?

manin22, 2019-03-10:
I have noticed the same issue on my system after Volumio upgrade to version 2.527 & Reset Album Art Cache. Same behavior after upgrade to current version 2.555.
Most of pictures disappeared and aren't displayed if browsing Music Library -> NAS. No png images (e.g. folder.png) at all and no folder.jpg named images are visible too.
I can only see non-default pictures if the file name is different then folder.jpg, e.g. joe-satriani.jpg or folder.jpeg.
The image appears back if the name of the file is renamed from folder.jpg to e.g. author.jpg. And of course after Album Art Cache reset.