volumio/Volumio2 #1663: all volumios suddenly fail to start--dependency issue

adipose, 2018-11-23:
All my volumios updated, and none would work, suddenly. It may have been after updating to 2.502.

When running manually, it complains about missing dependencies serve-favicon and cookie-parser. npm install did not install them, but after installing them manually, it began to work again.

Do we need to add those to the dependency list?

volumio, 2018-11-23:
touch /boot/factory_reset

This will revert to factory imsge

adipose, 2018-11-24:
Thanks--fixed.  After updating to 2.502, still fine.  It appears my image was a slightly modified one from "justboom."  I had thought it was just a theme change, but apparently there were other differences that broke it upon updating.

closed 2018-11-24 by adipose