volumio/Volumio2 #1665: Persistent MyVolumio banner is annoying

bjornarsu, 2018-11-25:
I just upgraded to 2.502, and while MyVolumio is probably a welcome new feature for some users, it's not something for everyone. I feel the "MyVolumio (Beta)" login/signup banner on top of the left side menu is annoying and should be possible to permanently remove/hide, as it kind of feels like an advertisement of sorts. Especially in what feels like a very personal device/space for me, that I use every day.

volumio, 2018-11-29:
I see what you mean. Never thought it could be annoying, since its something you can ignore as any other setting. We will see if there is a way to do it in a nice way, but can't promise anything ;)

closed 2018-11-29 by volumio

bjornarsu, 2019-09-19:
Any update on this?

volumio, 2019-09-19:
Volumio will always have myvolumio displayed there. If you don't login, nothing will be enabled on myvolumio.

DaveMDS, 2020-10-14:
Indeed, really annoying. And indeed seems like an intrusive advertiser
IMHO not in sync with a true open source software

volumio, 2020-10-14:
> Indeed, really annoying. And indeed seems like an intrusive advertiser
> IMHO not in sync with a true open source software

If the people using Volumio would donate to use it, then there would be no need for us to have a freemium service to keep it alive...

teebee2, 2021-01-12:
Yes, it totally sucks.  Volumio is fantastic!!  Like other pieces of essential audio equipment I would not mind paying for it, but I would like to pay a reasonable amount and I think the price structure is sadly very  inflated.  Someone got 'swimming pool goggles'