volumio/Volumio2 #1668: First couple of seconds of each track is always cut off with some DACs

davidcole1977, 2018-12-03:
This is my first experience with a Raspberry Pi based audio player and also with open source software, so please be nice if anything here doesn't adhere to community guidelines - give me feedback and I'll happily amend it :)

If no one is already working on a solution for this I'm happy to spend a little time on it and open a PR.

**Bug description:**

- When playing the first track in a playlist, the first 1-2 seconds of the track are always silent
- This happens every time. It is extremely consistent.
- It also happens if I press the "next" button to start playing the next track, or if the next track in the playlist is a different resolution to the previous track (eg. if track 1 is 44.1khz and track 2 is 96khz). And it happens if I play another track in the playlist by directly selecting that track.
- It does **not** seem to occur when I replay the existing track by pressing the "back" button
- It does **not** seem to occur when Volumio automatically plays the next track in the playlist after reach the end of the previous track
- It occurs when I play tracks from my library on an external USB storage
- It does **not** occur when using the Spotify connect plugin
- Changing the buffering settings in "Playback Options" does not have any effect on the problem.
- I do **not** have this problem with my other HiFi streamer, a Cyrus Stream X2 (which I hope to replace with my new DigiOne Signature & Volumio if this problem can be solved)
- This spoils my enjoyment enough and I doubt I'll continue using Volumio if it can't be fixed (but as mentioned, I might be able to help with that).

**Probably cause:**

- In my opinion, the silence is caused by the M-DAC taking a moment to 'lock on' to the new sampling frequency.
- Volumio does not appear to keep the output stream 'alive' when playing the first track in a playlist, or when using the "next" button to play a track, so the DAC always becomes 'unlocked' and needs to take a moment to lock on to the sampling frequency of the audio track, even if the sampling frequency hasn't changed from the previous track played.
- Volumio seems to behave differently when automatically playing the next track in the playlist - whatever it does to ensure gapless playback also seems to keep the signal 'alive' between tracks.
- The Volumio Spotify plugin seems to keep the output stream 'alive' at all times, no matter how I select the next track to play.
- The LCD display on the M-DAC confirms this behaviour - it provides very obvious information to tell me when the signal is being kept 'alive' (it continues to display the sampling frequency – eg. 44.1khz, but not the bit depth – eg. 16bit), or when it has become unlocked (it displays the word "Unlocked")

**Possible solution:**

- Aim to keep the output alive at all times when playing files from a storage drive (this might mean continuing to output a silent signal when no track is playing).
- We could assume the last played track is the 'correct' frequency to use as the basis for the 'alive' signal (this is what my Cyrus Stream X2 seems to do)
- We might also allow 1-2 seconds of silence in between tracks of different sampling frequencies.
- I'd tend to make this optional under "Playback Options", as this behaviour might not be optimal for all DACs 
- We should consider how the Spotify Connect plugin achieves an 'always alive' signal.

**My set up:**

- [Allo DigiOne Signature Player](https://www.allo.com/sparky/digione-signature-player.html), Raspberry Pi 3B+ and Volumio V2.502
- [Audiolab MDAC](http://www.audiolab.co.uk/product-detail.php?pid=13)
- Using SPDIF output
- External SSD storage drive

**Additional info:**

It looks like some other users have the same, or similar issue:

- https://volumio.org/forum/beginning-songs-cut-off-t2904.html
- https://volumio.org/forum/stutter-start-tracks-after-changing-dac-t7212.html

—

I am a professional Javascript/NodeJS developer and I'm happy to open a PR to fix this issue (my GitHib profile is extremely out of date and does not represent my current level of expertise). Before I start work on this:

- If anyone has some insight to offer regarding the solution, or if someone else has already worked on this issue, please let me know.
- I couldn't see a CONTRIBUTING.md file, so please let me know if there are any guidelines I should follow.
- Should I branch off `master`, `development`, or another branch?
- What are the guidelines around unit/integration/end-to-end tests?
- What is the newest compatible version of NodeJS?

Thank You!
David.

davidcole1977, 2018-12-03:
By the way, I think Volumio is superb in all other regards. For free, open source software it provides a brilliant experience. Well done to all of the the developers!

volumio, 2018-12-03:
Dear David,
what a nice introduction. Welcome to the Volumio project, IMHO your experience and attitude will surely be a great resource for the project. 
Yes, the issue has been affecting some users like you: under some circumstances, some DAC takes couple of seconds to adjust to the new sampling frequnency. Sometimes causing silence, sometimes causing pops.
Solution is not simple, but you're right that we should feed the DAC with an always on signal. 
Issues:
- If we keep the audio device "busy",other services like spotify or shairport may fail to initialize
- We should instruct mpd to play silence, but give away alsa playback in case some other service requires it.

Possible approach:
- You were right in saying that when tracks are played sequentially we don't have such issue. This is because we have a "prefetch" facility which adds tracks to queue when one finishes. This does not happen when you skip.
- We need therefore to fill those gaps with silence sound. And play it in the appropriate moment. Ideally, we should be able to synthethize this silent track based on the bitrate\bitdepth of the following track for which we want to avoid silence for.
- You're looking for the mpd plugin, which handles all mpd related tasks 
https://github.com/volumio/Volumio2/blob/master/app/plugins/music_service/mpd/index.js

Those are just my 2 cents, I am really looking forward to see what you think as well and if you have other elegant solutions in mind.

PS: You mentioned also CONTRIBUTING.md, we don't have it but feel free to add one so to help future devs join us.

To contribute:
- Fork the master branch
- Unfortunately we don't do testing, but simply because we lack time not because we don't want to. So, if you find a way to do TDD for the stuff you'll add, it will be very welcome.
- We are on node 8.11.1, but we'll eventually switch to 10 in the next couple of months
- Try to send small PRs, if its lot of things try to send many PRs

volumio, 2018-12-03:
Looking forward! Don't worry and take your time, in case feel free to ping me

davidcole1977, 2018-12-04:
@volumio I'm having a look to familiarise myself a little with some of the code. I'll don't want to ask too many questions when I can find the answer from the code, but one key question I have is around the use of modern Javascript (native promises, async/await, classes, Array.prototype.map(filter, reduce...), destructuring... in other words, all the lovely things we can now use in Node 8.x :)

I can see there are some things I would need to be careful of – eg. using native promises where the code expects kew.js promises. Other than these cases, are you happy to see some shiny modern Javascript in a PR?

volumio, 2018-12-04:
Yes, of course we want some new shiny js. You see lot of old school node just because we keep code coherence since we started 4 years ago. Only thing if you do some new node: comment when possible and keep the same functions structure (prototype)

davidcole1977, 2018-12-14:
Apologies for being silent the past week. I’ve not written any code yet, but have been familiarising myself the Volumio repository and wiki etc.

My thoughts on the solution:

- As suggested, I think something that would work is to add a short length of silence – matching the resolution of the next track – begire playing a new track. This could be done using MPD’s internal queue, much as the prefetch works.
- This should be optional, as users who aren’t affected by the issue may prefer the additional responsiveness.
- I think a more sophisticated solution may be possible, but it’s complicated by being able to queue up tracks from different plugins/services in the Volumio queue. For now I think it will keep things a lot simpler if we don’t make changes to how the main controller / queue works.
- The MPD plugin will need to know the resolution of the tracks in the queue (well, the tracks that will be played by MPD anyway) and I think it probably makes sense to get that information when the track is added to the queue.
- To start with it’s probably easier to have a selection of ‘silence’ WAV or FLAC files already prepared with the most common resolutions. We can then look at whether a clever way of generating the silence is needed.
- I’d quite like to add some additional debugging that will let us see the status of the MPD queue at any time, perhaps an addition to the ‘/dev’ route.
- The fix could easily be done as a number of small PRs, with the final ‘release’ being enabled by adding the configuration option to the UI.

@volumio does that all sound reasonable?

I’m waiting on a spare micro SD card for my raspberry pi so I can have a separate development setup. I’ll start to make the code changes then.

volumio, 2018-12-17:
Dear David, yes that sounds like a perfect solution! Looking forward !

davidcole1977, 2019-01-25:
@Garysycheung no worries :)

I got a bit side tracked and am now just starting to work on a solution for this. However, I'm afraid it's likely to work only for music played from your local storage device and is not likely to affect Spotify Connect, or other plugins.

Sorry to disappoint you. Once we have a fix for local music files and I know the codebase a bit better, then it's possible I might have a better idea how a fix could be applied to other plugins. At this time, Spotify Connect isn't something I'm planning to look at, but it's something I also use a lot, so if I experience the same problems then there will certainly be some motivation for me ;)

Garysycheung, 2019-01-25:
Hi @davidcole1977 , thank you very much for your reply😊 , I have done some more research last night and have found that there are quite a few people suffering from the issue:

https://volumio.org/forum/always-pcm-signal-dac-t2159-10.html

https://volumio.org/forum/start-selected-song-cut-off-t11255.html

Anyway I have done some test and try to post to the Volumio forum with my result but my post never appear😡. 

For sure that the 'cut off' during the first few second of the song is causing by the PCM 'switch off' and on between tracks. 

I have done some experience by switching back to use the Chromecast audio (with Toslink), and have observed that the Chromecast audio did actually keep the PCM connection always on, hence this issue never happen.

Further more, I have done a control experiment by using a CD player with coaxial to drive the Audiolab MDAC+, and appeared that it did not 'switch off' the PCM connection between tracks..... 
This experience proven that it is not cause by the difference between Coaxial and Toslink.

Sadly I can't do any coding myself 😓 , but I would suggested that the issue can be fix if there is a way to make the volumio to keep the PCM connection to become 'always on' after the pi been powered on. 
I do not see any valid reason why it have to 'switch off' the PCM connection at any time, If other device could. Perhaps this is the right direction ?

Many thanks.

davidcole1977, 2019-01-30:
@Garysycheung I reckon you're thinking along the correct lines, but unfortunately the solution I'm working on is a simple solution that won't keep the connection "always on", but instead will work by playing maybe one or two seconds of silence before starting to play the track – and this would only work for music files in someone's local collection (eg. on a hard disk drive or USB stick), not Spotify.

I had thought to aim for a more sophisticated solution, but honestly it's something I might not succeed with at this time. The way I see it, it's better to work on something simple that could help some people, rather than a complex solution that might never get finished. I'm sorry that's not much help for you :(

davidcole1977, 2019-01-30:
@volumio sorry I've been silent about this issue for a long time. There was Christmas, then being ill with flu, family commitments and then of course I got distracted by starting to create a new Volumio UI when I should have been working on this ;)

Anyway, I have started to work on this now. I put together a very simple proof of concept that simply adds a 3 seconds/44.1khz/16bit silence track to the start of the MPD queue any time you play a new track, or skip to the next track. Something like this:

```javascript
ControllerMpd.prototype.clearAddPlayTrack = function (track) {
    // ...

    if(track.uri.startsWith('cue://'))
    {
        // ...
    }
    else{
       // ...

        // Clear the queue, add the first track, and start playback
        // ...

        return self.sendMpdCommand('stop',[])
            .then(function()
            {
                return self.sendMpdCommand('clear',[]);
            })
            .then(function()
            {
                return self.sendMpdCommand('add "' + silenceUri + '"',[]);
            })
            .then(function()
            {
                return self.sendMpdCommand('add "'+safeUri+'"',[]);
            })
            .then(function()
            {
                return self.sendMpdCommand('play',[]);
            });
    }
};
```

I listened to an album that previously had some big problems and the sound at the start of each track was perfect :) 

However, one problem I noticed is that if I poke the user interface much then it's possible for the wrong track to play – I need to investigate more, but I imagine adding the extra track to the MPD queue 'confuses' Volumio in some circumstances.

I'll continue to work on it and will update you when there is some progress – same caveat applies however: no promises about how long it will take me.

Garysycheung, 2019-01-30:
Hi @davidcole1977 , Thank you very much for your information, and no worry :) My bad, I should have give you some update as well. I have also tried to look into the issues, and have investigated the MPD conf file with the information from this link (I am sure that you have been there before) :

 https://www.musicpd.org/doc/html/user.html

However, I can't anything that would resolve the issue after some trial an error. Hence, I try another solution by ditching the Volumio, and tried a few other players. For those I have tried so far, they didn't have this cut off issue. Seriously @volumio , I don't understand why other players could have resolve the issue but not the Volumio. 

Anyway, I am now running the piCorePlayer, with the Logitech Music Server and Squeezelite, then within the LMS, I use the 'Spotty' plugin which provide Spotify and Spotify connect function... it keep the PCM signal and NEVER cut off...... so everything is FINE, in a way that it solved the problem. 

Forgot to say, the Spotify plugin in Volumio didn't work for me either, so the piCorePlayer hit two birds with a single stone...

I then realised that Volumio also have the same LMS and Squeezelite plugin so I was looking at this direction -> IF I run the LMS in Volumio, then I could ALSO solve the problem?! 

However, the LMS server in my Volumio, DIDN'T work as well! google it around and found that there are another bug / issue...... 

I am making a stupid assumption that it will behaving the same as the piCorePlayer without cut off, however, I can't try it with mine. 

You may want to have a try and see if it work for you? 
@volumio can you confirm that the LMS pliugin will still have the cut off? 

Saying that I am using another player for Spotify connect at the moment, I still, havent give up the Volumio :)  and will try to look into it.... 

@volumio are there any plan to resolve the LMS plugin and Spotify issue?

Many thanks :)

davidcole1977, 2019-01-30:
Note: ffmpeg could be useful for programmatically creating a range of silent audio files: 
https://stackoverflow.com/questions/32017827/ffmpeg-create-silent-ogg-audio-file?rq=1

davidcole1977, 2019-02-11:
I'm continuing to spend some time on this each week, but it's proving to be problematic. There can be a kind of race condition in which the prefetch is triggered when you attempt to play a new track. It is taking me some time just to work out what is actually happening before I can attempt to fix it...

volumio, 2019-02-13:
@davidcole1977 feel free to ask if in need!
PS: IMHO this is probably one of the most difficult tasks one could choose in Volumio ;)

davidcole1977, 2019-02-13:
@volumio thanks – I’ll definitely want some feedback once I’ve got something working. The solution I’m working on is crude and simple – the difficult part is making sure I don’t introduce bugs through lack of understanding how the application fits together, which I’m sure you can help with. But for now the best way for me to get to know the application is to see what breaks when I make a change ;)

yog-sothoth0, 2020-05-11:
Hi all,

I have this same problem with the beginning of songs being cut off in Volumio (and not with other players, as previously described.  My setup is a bit different -- I'm running Volumio on an Intel NUC and playing the music through the HDMI out directly into the DAC in my Rotel 1572. I love Volumio so far, and even made a separate one (On a Lemvo Thinkcenter) to give to my brother.

While I don not have any Java programming experience, I do have 35 years of Unix/Solaris/Linux etc. systems engineering experience (I would still be working bu health dissonance took it's toll on me); I can maneuver through the inside of the system like a seal frolicking in the water.  If I can help with any sort of testing for all of your excellent work, I'd love to be able to contribute any little bit I might.  I have plenty of spare laptops and Thinkcenters that I could destroy and create for a testing environment and whim.

Please let me know if I can do anything withing my skill set to help.  And I'm not against learning new material.

bma25, 2020-08-02:
@davidcole1977 Are you still working on this?