volumio/Volumio2 #1686: USB Audio quirks for Kernel 4.14.90

gkkpch, 2018-12-24:
I have a patch, assembled from from 4.18.y and 4.19.y, which brings DSD Direct capable device support to the level of the last stable kernel.. 
This patch has been tested with kernel 4.14.90 on cubox-i and works perfectly.
I added two kernel print messages for XMOS/Thesycon/Mytek devices, which will show if and which altsetting is DSD Direct capable.
These patches are **not** in the PI kernel, so if we want to use them, we have to cokpile our own.

diff --git a/sound/usb/card.h b/sound/usb/card.h
index ed87cc83..2f4801cd 100644
--- a/sound/usb/card.h
+++ b/sound/usb/card.h
@@ -32,6 +32,7 @@ struct audioformat {
 	struct snd_pcm_chmap_elem *chmap; /* (optional) channel map */
 	bool dsd_dop;			/* add DOP headers in case of DSD samples */
 	bool dsd_bitrev;		/* reverse the bits of each DSD sample */
+	bool dsd_raw;			/* altsetting is raw DSD */
 };
 
 struct snd_usb_substream;
diff --git a/sound/usb/format.c b/sound/usb/format.c
index 2c44386e..3e5f38d7 100644
--- a/sound/usb/format.c
+++ b/sound/usb/format.c
@@ -63,9 +63,11 @@ static u64 parse_audio_format_i_type(struct snd_usb_audio *chip,
 		sample_width = fmt->bBitResolution;
 		sample_bytes = fmt->bSubslotSize;
 
-		if (format & UAC2_FORMAT_TYPE_I_RAW_DATA)
+		if (format & UAC2_FORMAT_TYPE_I_RAW_DATA) {
 			pcm_formats |= SNDRV_PCM_FMTBIT_SPECIAL;
-
+			/* flag potentially raw DSD capable altsettings */
+			fp->dsd_raw = true;
+		}
 		format <<= 1;
 		break;
 	}
diff --git a/sound/usb/quirks.c b/sound/usb/quirks.c
index ad14d6b7..65a38b64 100644
--- a/sound/usb/quirks.c
+++ b/sound/usb/quirks.c
@@ -1355,20 +1355,44 @@ u64 snd_usb_interface_dsd_format_quirks(struct snd_usb_audio *chip,
 
 	/* XMOS based USB DACs */
 	switch (chip->usb_id) {
-	case USB_ID(0x20b1, 0x3008): /* iFi Audio micro/nano iDSD */
+	case USB_ID(0x1511, 0x0037): /* AURALiC VEGA */
+	case USB_ID(0x20b1, 0x0002): /* Wyred 4 Sound DAC-2 DSD */
+	case USB_ID(0x20b1, 0x2004): /* Matrix Audio X-SPDIF 2 */
 	case USB_ID(0x20b1, 0x2008): /* Matrix Audio X-Sabre */
 	case USB_ID(0x20b1, 0x300a): /* Matrix Audio Mini-i Pro */
 	case USB_ID(0x22d9, 0x0416): /* OPPO HA-1 */
+	case USB_ID(0x22d9, 0x0436): /* OPPO Sonica */
+	case USB_ID(0x22d9, 0x0461): /* OPPO UDP-205 */
+	case USB_ID(0x2522, 0x0012): /* LH Labs VI DAC Infinity */
 	case USB_ID(0x2772, 0x0230): /* Pro-Ject Pre Box S2 Digital */
 		if (fp->altsetting == 2)
 			return SNDRV_PCM_FMTBIT_DSD_U32_BE;
 		break;
 
+	case USB_ID(0x152a, 0x85de): /* SMSL D1 DAC */
+	case USB_ID(0x16d0, 0x09dd): /* Encore mDSD */
+	case USB_ID(0x0d8c, 0x0316): /* Hegel HD12 DSD */
+	case USB_ID(0x16b0, 0x06b2): /* NuPrime DAC-10 */
+	case USB_ID(0x16d0, 0x0733): /* Furutech ADL Stratos */
+	case USB_ID(0x16d0, 0x09db): /* NuPrime Audio DAC-9 */
+	case USB_ID(0x1db5, 0x0003): /* Bryston BDA3 */
 	case USB_ID(0x20b1, 0x000a): /* Gustard DAC-X20U */
+	case USB_ID(0x20b1, 0x2005): /* Denafrips Ares DAC */
 	case USB_ID(0x20b1, 0x2009): /* DIYINHK DSD DXD 384kHz USB to I2S/DSD */
 	case USB_ID(0x20b1, 0x2023): /* JLsounds I2SoverUSB */
+	case USB_ID(0x20b1, 0x3021): /* Eastern El. MiniMax Tube DAC Supreme */
 	case USB_ID(0x20b1, 0x3023): /* Aune X1S 32BIT/384 DSD DAC */
+	case USB_ID(0x20b1, 0x302d): /* Unison Research Unico CD Due */
+	case USB_ID(0x20b1, 0x307b): /* CH Precision C1 DAC */
+	case USB_ID(0x20b1, 0x3086): /* Singxer F-1 converter board */
+	case USB_ID(0x22d9, 0x0426): /* OPPO HA-2 */
+	case USB_ID(0x22e1, 0xca01): /* HDTA Serenade DSD */
+	case USB_ID(0x249c, 0x9326): /* M2Tech Young MkIII */
 	case USB_ID(0x2616, 0x0106): /* PS Audio NuWave DAC */
+	case USB_ID(0x2622, 0x0041): /* Audiolab M-DAC+ */
+	case USB_ID(0x27f7, 0x3002): /* W4S DAC-2v2SE */
+	case USB_ID(0x29a2, 0x0086): /* Mutec MC3+ USB */
+	case USB_ID(0x6b42, 0x0042): /* MSB Technology */
 		if (fp->altsetting == 3)
 			return SNDRV_PCM_FMTBIT_DSD_U32_BE;
 		break;
@@ -1411,5 +1435,22 @@ u64 snd_usb_interface_dsd_format_quirks(struct snd_usb_audio *chip,
 			return SNDRV_PCM_FMTBIT_DSD_U32_BE;
 	}
 
+	/* Mostly generic method to detect many DSD-capable implementations -
+	 * from XMOS/Thesycon
+	 */
+	switch (USB_ID_VENDOR(chip->usb_id)) {
+	case 0x20b1:  /* XMOS based devices */
+	case 0x152a:  /* Thesycon devices */
+	case 0x25ce:  /* Mytek devices */
+		printk(KERN_DEBUG "XMOS USB Audio device: checking altsetting %d\n",fp->altsetting);
+		if (fp->dsd_raw) {
+			printk(KERN_DEBUG "DSD Direct capable!\n");
+			return SNDRV_PCM_FMTBIT_DSD_U32_BE;
+		}
+		break;
+	default:
+		break;
+
+	}
 	return 0;
 }

volumio, 2018-12-24:
What do you suggest? Shall we go and compile all kernels?

gkkpch, 2018-12-24:
We do most of them already, except PI, Cubox and Asus (when we get back to normal after the Tinker clock issue has been fixed).
With Asus we can get patches integrated fairly easy, with Cubox I rely on Armbian (might actually use that for other main platforms. Armbian’s build method offers the opportunity to patch ourselves.
Leaves the PI. 
Not a lot of work after doing that for the first time, perhaps @macmpi is willing to maintain that (there is a weekly update for 4.14.y).

But this also reminds me of one thing I have not done consistently: document it properly.

macmpi, 2018-12-26:
@gkkpch you may want to suggest the patch to @pelwell and crew at https://github.com/raspberrypi/linux/issues: they might cherry-pick into next 4.14.x, as this is potentially beneficial to many.

volumio, 2018-12-26:
Great idea! 

And I also had another one for the pi kernel: for the USB quirks to be added, why don't we just rebuild the ko file and add it on top of the official pi kernel?

pelwell, 2018-12-26:
Go for it. I don't think we'll be on 4.14 much longer, but anything which is essentially a back port is usually accepted very quickly.

gkkpch, 2018-12-27:
@pelwell / @macmpi 
I followed your advice and created a PR with it.

gkkpch, 2018-12-28:
I’ll close this one, it has just been merged into the PI kernel 4.14.y as #2790.

closed 2018-12-28 by gkkpch