volumio/Volumio2 #1687: REST API calls take too long

deeplook, 2018-12-25:
I was playing with the REST API for the first time and noticed that making basically any call takes pretty much always five seconds to return. This renders the API quite unusable for anything like experimental player interfaces in other apps. I don't think this should be expected within the same network. Is there any reason for this? I'm running system version 2.513.

volumio, 2018-12-26:
That's not the timing we usually see: they are pretty fast.
Which call are you sending?

deeplook, 2018-12-26:
Basically any calls, here are three basic ones. I've retested now directly after rebooting my Raspberry Pi 2 (I believe). But rebooting shows no effect on the response time.
[image]