volumio/Volumio2 #1690: Cannot Add Single Songs from NAS to Playlist. Cannot Save Web Radio to Favorites.

thomastech, 2018-12-30:
**Background Info:**
Version 2.513 on a RPI3 Model B, 16GB SD.  
I've re-flashed a new SD card and started over. The problem remains.
Plugins: Autostart 1.1.2, 80s80sRadio 1.0.0, Tunein Radio 1.1.0, Youtube Playback 0.0.9, Backup & Restore 0.7.2, GPIO Buttons 0.8.1

**Problem Summary:**
1. Cannot add single songs from "MediaServers->NAS" to Playlist. Adding songs to Playlist from "MusicLibrary->NAS" is Ok.

2. Cannot save web radio stations to favorites. Specifically I tried saving "Tunein Radio" and "80's80's Radio" web stations to "Web Radio->Radio Favourites". I have no problems saving "Volumio Selection" stations to Radio Favourites.

Another user has similar issues adding single songs. See:
https://volumio.org/forum/can-add-single-songs-playlist-t11267.html

BTW, after deleting an entry from "Web Radio->Favorite Radios" the web page reverts to the main "Favorites" page. It should stay on "Web Radio->Favorite Radios".

- Thomas

xipmix, 2019-01-01:
Tested with 2.513, same behaviour observed - very little to go on in the logs.

xipmix, 2019-01-01:
Workaround: adding a track to the play queue is supported, so I tried adding a track to the queue and saving the queue as a playlist. That much works.

But when I try to use that playlist I hit problems (version 2.513).
 * clear queue
 * Browse->Playlists->menu for playlist3 -> add to queue
```
Jan 01 05:31:02 volumio volumio[13156]: info: Enqueue Playlist3
Jan 01 05:31:02 volumio volumio[13156]: info: CoreCommandRouter::volumioAddQueueItems
Jan 01 05:31:02 volumio volumio[13156]: info: CoreStateMachine::addQueueItems
Jan 01 05:31:02 volumio volumio[13156]: info: CorePlayQueue::addQueueItems
Jan 01 05:31:02 volumio volumio[13156]: info: Adding Item to queue: http://192.168.x.y:9000/disk/NON-DLNA/music/O0$1$7I13/Best%20Friend.flac
Jan 01 05:31:02 volumio volumio[13156]: info: Exploding uri http://192.168.x.y:9000/disk/NON-DLNA/music/O0$1$7I13/Best%20Friend.flac in service upnp_browser
Jan 01 05:31:02 volumio volumio[13156]: error:  message=Did not get expected response from server:{"HTML":{"HEAD":[{"TITLE":[{"_":"404 Not Found"}]}],"BODY":[{"_":"The requested URL was not found on this server.","H1":[{"_":"Not Found"}]}]}}, stack=Error: Did not get expected response from server:{"HTML":{"HEAD":[{"TITLE":[{"_":"404 Not Found"}]}],"BODY":[{"_":"The requested URL was not found on this server.","H1":[{"_":"Not Found"}]}]}}
Jan 01 05:31:02 volumio volumio[13156]: at /volumio/app/plugins/music_service/upnp_browser/dlna-browser.js:140:24
Jan 01 05:31:02 volumio volumio[13156]: at Parser.<anonymous> (/volumio/node_modules/xml2js/lib/parser.js:303:18)
Jan 01 05:31:02 volumio volumio[13156]: at emitOne (events.js:116:13)
Jan 01 05:31:02 volumio volumio[13156]: at Parser.emit (events.js:211:7)
Jan 01 05:31:02 volumio volumio[13156]: at SAXParser.onclosetag (/volumio/node_modules/xml2js/lib/parser.js:261:26)
Jan 01 05:31:02 volumio volumio[13156]: at emit (/volumio/node_modules/sax/lib/sax.js:624:35)
Jan 01 05:31:02 volumio volumio[13156]: at emitNode (/volumio/node_modules/sax/lib/sax.js:629:5)
Jan 01 05:31:02 volumio volumio[13156]: at closeTag (/volumio/node_modules/sax/lib/sax.js:889:7)
Jan 01 05:31:02 volumio volumio[13156]: at SAXParser.write (/volumio/node_modules/sax/lib/sax.js:1436:13)
Jan 01 05:31:02 volumio volumio[13156]: at Parser.exports.Parser.Parser.parseString (/volumio/node_modules/xml2js/lib/parser.js:322:31)
Jan 01 05:31:02 volumio volumio[13156]: at Parser.parseString (/volumio/node_modules/xml2js/lib/parser.js:5:59)
Jan 01 05:31:02 volumio volumio[13156]: at Object.exports.parseString (/volumio/node_modules/xml2js/lib/parser.js:354:19)
Jan 01 05:31:02 volumio volumio[13156]: at IncomingMessage.<anonymous> (/volumio/app/plugins/music_service/upnp_browser/dlna-browser.js:93:15)
Jan 01 05:31:02 volumio volumio[13156]: at emitNone (events.js:111:20)
Jan 01 05:31:02 volumio volumio[13156]: at IncomingMessage.emit (events.js:208:7)
Jan 01 05:31:02 volumio volumio[13156]: at endReadableNT (_stream_readable.js:1064:12)
```

xipmix, 2019-01-02:
> I'm new to Volumio and unfamiliar with its support community.

The main developer is @volumio, there seem to be a couple of other people who have commit access (not me). The forums might be a good place to learn about this - https://volumio.org/forum/

thomastech, 2019-01-02:
Thank you for the developer info. Hopefully @volumio lends a hand with this issue.