volumio/Volumio2 #1691: Favorite web radio gone

cmoulliard, 2018-12-31:
### Info

Version: 2.513

### Description

I have added new web favorites radio within the "/data/favourites/my-web-radio" file but when I reboot the Pi3 then Volumio don't show them anymore. Where should I then save them to be able to see the radio under the UI location - "My Web Radios" ?

[image]

cmoulliard, 2018-12-31:
After looking to the [code](https://github.com/volumio/Volumio2/blob/master/app/plugins/music_service/webradio/index.js#L707),

```
 for (var i in data) {
            var ithdata = data[i];
            var song = {
                service: 'webradio',
                type: 'mywebradio',
				artist: '',
				album: '',
                title: ithdata.name,
                uri: ithdata.uri,
                icon: 'fa fa-microphone'
            };
```
I have discovered that it is only needed to keep as data the `name` and `uri`. So I have updated my json file as asuch and now the content is not longer deleted when I reboot and when volumio get the favourite radio list

```
 cat /data/favourites/my-web-radio
[
  {"name":"Classic21","uri":"https://radios.rtbf.be/classic21-128.mp3"},
  {"name":"Pure","uri":"https://radios.rtbf.be/pure-128.mp3"},
  {"name":"Nostalgie","uri":"http://nostalgiepremium.ice.infomaniak.ch/nostalgiepremium-128.mp3"},
  {"name":"Vivacite Charleroi","uri":"https://radios.rtbf.be/vivacharleroi-128.mp3"},
  {"name":"La Premiere","uri":"https://radios.rtbf.be/laprem1ere-128.mp3"},
  {"name":"NRJ","uri":"http://nrjbe.ice.infomaniak.ch/nrjbe-high.mp3"},
  {"name":"Bel RTL","uri":"http://audiostream.rtl.be/belrtl128"},
  {"name":"Contact","uri":"http://audiostream.rtl.be/contactfr192"},
  {"name":"Musique 3","uri":"https://radios.rtbf.be/musiq3-128.mp3"},
  {"name":"RTL2","uri":"http://icecast.rtl.fr:80/rtl2-1-44-128"},
  {"name":"Europe1","uri":"http://e1-live-mp3-128.scdn.arkena.com/europe1.mp3"},
  {"name":"Noel - Vivacite","uri":"https://radios.rtbf.be/wr-viva-xmas-128.mp3"},
  {"name":"Noel - Nostalgie","uri":"http://nostalgielastory.ice.infomaniak.ch/nostalgielastory-128.mp3"},
  {"name":"Nostalgie 80","uri":"http://nostalgie80.ice.infomaniak.ch/nostalgie80-128.mp3"},
  {"name":"Nostalgie 90","uri":"http://nostalgie90.ice.infomaniak.ch/nostalgie90-128.mp3"},
  {"name":"Nostalgie Love","uri":"http://nostalgielove.ice.infomaniak.ch/nostalgielove-128.mp3"},
  {"name":"Nostalgie - Chanson Francaise","uri":"http://nostalgiechansonfrancaise.ice.infomaniak.ch/nostalgiechansonfrancaise-128.mp3"}
]
```

xipmix, 2019-01-01:
Thank you for the report. 

If I understand correctly, you noticed that extra elements in the /data/favourites/my-web-radio.json file caused the system to delete that file after a reboot. Or is it the file exists but is empty after reboot?

Now that you have a working file, could  you add ```artist: '', album: '',``` to one of the items in the /data/favourites/my-web-radio.json file and check if the content is deleted after a reboot ? Obviously make a backup copy of the file first...

cmoulliard, 2019-01-01:
> Or is it the file exists but is empty after reboot?

I will make a new test to verify if the file exists and is empty after reboot in order to answer to your question like also to to your next point where I will add `artist: '', album: ''`, to one of the items in the /data/favourites/my-web-radio.json file

cmoulliard, 2019-01-01:
I did multiple reboot today and noticed that again the list only contained a few items and  not all. 

Here is the file - ` /data/favourites/my-web-radio` containing the items added some days ago using the UI which is incomplete.

```
[{"name":"Classic21","service":"webradio","uri":"https://radios.rtbf.be/classic21-128.mp3"},{"name":"La Premiere","service":"webradio","uri":"https://radios.rtbf.be/laprem1ere-128.mp3"},{"name":"Musique 3","service":"webradio","uri":"https://radios.rtbf.be/musiq3-128.mp3"},{"name":"Pure","service":"webradio","uri":"https://radios.rtbf.be/pure-128.mp3"},{"name":"Nostalgie","service":"webradio","uri":"http://nostalgiepremium.ice.infomaniak.ch/nostalgiepremium-128.mp3"},{"name":"Vivacite Charleroi","service":"webradio","uri":"https://radios.rtbf.be/vivacharleroi-128.mp3"},{"name":"RTL2","service":"webradio","uri":"http://icecast.rtl.fr:80/rtl2-1-44-128"}]
```

I have update it to include all the items like also `artist: '', album: ''` as you mentioned and rebooted again the Pi3. After doing 2 reboots, the file's content looks good now.

```
[
  {"artist":"","album":"","name":"Classic21","uri":"https://radios.rtbf.be/classic21-128.mp3"},
  {"artist":"","album":"","name":"Pure","uri":"https://radios.rtbf.be/pure-128.mp3"},
 {"artist":"","album":"","name":"Nostalgie","uri":"http://nostalgiepremium.ice.infomaniak.ch/nostalgiepremium-128.mp3"},
  {"artist":"","album":"","name":"Vivacite Charleroi","uri":"https://radios.rtbf.be/vivacharleroi-128.mp3"},
  {"artist":"","album":"","name":"La Premiere","uri":"https://radios.rtbf.be/laprem1ere-128.mp3"},
  {"artist":"","album":"","name":"NRJ","uri":"http://nrjbe.ice.infomaniak.ch/nrjbe-high.mp3"},
  {"artist":"","album":"","name":"Bel RTL","uri":"http://audiostream.rtl.be/belrtl128"},
  {"artist":"","album":"","name":"Contact","uri":"http://audiostream.rtl.be/contactfr192"},
  {"artist":"","album":"","name":"Musique 3","uri":"https://radios.rtbf.be/musiq3-128.mp3"},
  {"artist":"","album":"","name":"RTL2","uri":"http://icecast.rtl.fr:80/rtl2-1-44-128"},
  {"artist":"","album":"","name":"Europe1","uri":"http://e1-live-mp3-128.scdn.arkena.com/europe1.mp3"},
  {"artist":"","album":"","name":"Noel - Vivacite","uri":"https://radios.rtbf.be/wr-viva-xmas-128.mp3"},
  {"artist":"","album":"","name":"Noel - Nostalgie","uri":"http://nostalgielastory.ice.infomaniak.ch/nostalgielastory-128.mp3"},
  {"artist":"","album":"","name":"Nostalgie 80","uri":"http://nostalgie80.ice.infomaniak.ch/nostalgie80-128.mp3"},
  {"artist":"","album":"","name":"Nostalgie 90","uri":"http://nostalgie90.ice.infomaniak.ch/nostalgie90-128.mp3"},
  {"artist":"","album":"","name":"Nostalgie Love","uri":"http://nostalgielove.ice.infomaniak.ch/nostalgielove-128.mp3"},
  {"artist":"","album":"","name":"Nostalgie - Chanson Francaise","uri":"http://nostalgiechansonfrancaise.ice.infomaniak.ch/nostalgiechansonfrancaise-128.mp3"}
]
```

cmoulliard, 2019-01-02:
The list of the radio as defined under `/data/favourites/my-web-radio` is deleted when we reboot it using the UI. I will continue my invetigations as this problem is really annoying and should be labelised under github as `Bug` @xipmix

cmoulliard, 2019-01-02:
I confirm that when I restart volumio using the UI -> `shutdown -> restart`, then the list of my `/data/favourites/my-web-radio` json file is reduced to the list of the Web Radio added using the UI

Questions : 
- Where are stored the Web Radios added using the UI of volumio ?
- Why the file `/data/favourites/my-web-radio` is replaced with the UI saved web radios when we reboot the UI ?

xipmix, 2019-01-02:
> Where are stored the Web Radios added using the UI of volumio ?

I think it is either /data/favourites/radio-favourites or /data/favourites/my-web-radio

cmoulliard, 2019-01-02:
The file ` /data/favourites/radio-favourites` only contains one entry for me and is used for a different purpose as they will appear within the UI under the line `fa heart - Favorite radios`

```
[{"icon":"fa-microphone","service":"webradio","title":"classic21-128.mp3","uri":"https://radios.rtbf.be/classic21-128.mp3"}]
```

The `data/favourites/my-web-radio` file is used to display the list of the radios under the `fa heartbeat - My Web radio`.

I have started to look to the code of the UI to determine where the radios that we add using the UI and this screen are stored as apparently this list overwrittes the list of the file `data/favourites/my-web-radio` (that you can create manually) when the volumio is restarted using the UI

[image]

xipmix, 2019-01-02:
I suspect the issue lies in the backend but I can't see where. There are no ```unlink()``` calls that seem unwarranted. So if you find something in the ui code, great.

xipmix, 2019-01-02:
Here's something a bit odd in app/playlistManager.js
```
PlaylistManager.prototype.commonAddToPlaylist = function (folder, name, service, uri, title) 
...
    fs.exists(filePath, function (exists) {                                     
        var fileDefer=libQ.defer();                                             
                                                                                
                                                                                
        if (!exists) {                                                          
                //console.log("Setting default value for favourite "+folder+" "+name)
                //fileDefer=self.saveJSONFile(folder, name, playlist)           
            fs.writeJson(folder + name, playlist, function (err) {              
                if (err)                                                        
                    fileDefer.reject(new Error());                              
                else fileDefer.resolve();                                       
            })                                                                  
                                                                                
        } else fileDefer.resolve();                                             
```
which seems to work ok. But here
```
PlaylistManager.prototype.addToMyWebRadio = function (service, radio_name, uri) {
...
    fs.exists(filePath, function (exists) {                                     
        var fileDefer;                                                          
        if (!exists) {                                                          
            playlist.push({service: service, name: radio_name, uri: uri});      
            fileDefer=self.saveJSONFile(folder, name, playlist);                
        } else fileDefer=libQ.resolve();                                        
                                                                                
```
I wonder why the saveJSONFile method was commented out in the other function...

But none of this should be getting executed because we are in the 'file exists' case.

cmoulliard, 2019-01-02:
Code of `addToMyWebRadio ` should be perhaps changed to throw as `commonAddToPlaylist` an error if we can't write to the json file.

What I don't understand personally is what happens after the content of the file `/data/favourites/my-web-radio` has been loaded when volumio starts and music_service/webradio plugin has been called. Is the content overwritten by the UI ? Who can help us on that ? 

https://github.com/volumio/Volumio2/blob/master/app/plugins/music_service/webradio/index.js#L680-L719

```
 var promise = self.commandRouter.playListManager.getMyWebRadioContent()
    promise.then(function (data) {
        //console.log(data);
        data.sort(
            function (a, b) {
                return a.name.toLowerCase().localeCompare(b.name.toLowerCase());
            }
        );

        var response = {
            "navigation": {
                "lists": [
                    {
                        "availableListViews": [
                            "list"
                        ],
                        "items": [

                        ]
                    }
                ],
                "prev": {
                    "uri": "radio"
                }
            }
        };

        for (var i in data) {
            var ithdata = data[i];
            var song = {
                service: 'webradio',
                type: 'mywebradio',
				artist: '',
				album: '',
                title: ithdata.name,
                uri: ithdata.uri,
                icon: 'fa fa-microphone'
            };

            response.navigation.lists[0].items.push(song);
```

Bering-Sea, 2019-11-22:
Anyone have a solution for this yet? 

Thanks

volumio, 2019-11-22:
This is fixed in the latest version

cmoulliard, 2019-12-21:
> This is fixed in the latest version

Which version ? @volumio. I did a test today with version `2.688` and problem is still there

xipmix, 2019-12-22:
Latest for pi is 2.692, released 19 December.

cmoulliard, 2019-12-22:
Thanks. Then the problem is not yet solved

cmoulliard, 2019-12-28:
> This is fixed

What is the link about the git commit ?

cmoulliard, 2020-01-04:
I know why the favorites radio are gone. If you create a `Myvolumio.org` account and log in using Volumio UI, then every time we will restart / reboot the PiBox, volumio will pick up the playlist and favorites from `Myvolumio.org`

Questions:
- What is exactly sync between volumio and myvolumio account : playlist, favorite radios, .... ?
- Can we sync "My Web radios" or only "Favorite radios" ?
- Can we manage what we sync ? Is there a file which is used by volumio to sync ?
- When sync will take place and how often ?

cmoulliard, 2020-11-07:
> I know why the favorites radio are gone. If you create a `Myvolumio.org` account and log in using Volumio UI, then every time we will restart / reboot the PiBox, volumio will pick up the playlist and favorites from `Myvolumio.org`

In fact, the problem is still there even if we are not logged to `myvolumio.org` using version `2.692`

[image]

After a reboot, this file just contains a few radio
```
cat  /data/favourites/my-web-radio
[{"name":"Classic21","service":"webradio","uri":"https://radios.rtbf.be/classic21-128.mp3"},{"name":"La Premiere","service":"webradio","uri":"https://radios.rtbf.be/laprem1ere-128.mp3"},{"name":"Musique 3","service":"webradio","uri":"https://radios.rtbf.be/musiq3-128.mp3"},{"name":"Pure","service":"webradio","uri":"https://radios.rtbf.be/pure-128.mp3"},{"name":"Nostalgie","service":"webradio","uri":"http://nostalgiepremium.ice.infomaniak.ch/nostalgiepremium-128.mp3"},{"name":"Vivacite Charleroi","service":"webradio","uri":"https://radios.rtbf.be/vivacharleroi-128.mp3"},{"name":"RTL2","service":"webradio","uri":"http://icecast.rtl.fr:80/rtl2-1-44-128"}]
```

The only workaround is to scp the file from my macbook for the moment after every reboot of pi+volumio
```
 scp ~/code/ch007m/my-volumio-dev/radio-playlist/my-web-radio volumio@192.168.1.100:/data/favourites/
```

cmoulliard, 2020-11-07:
Fixed since version 2.834. I upgraded my volumio, reboot it after adding my web radios and they are not gone anymore

closed 2020-11-07 by cmoulliard