volumio/Volumio2 #1693: [question] How can we start automatically the player when volumio reboots

cmoulliard, 2019-01-01:
### Question

How can we start automatically the player when volumio reboots ? I have created a bash script which is called by a crontab rule `@reboot` but nothing happens. 

Remark : I can execute manually the bash script containing a curl command to call `playplaylist` and that works. 

### Steps

- Bash script file
```
volumio@volumio:~$ cat start-playlist.sh 
#/bin/sh
 
volumio=localhost:3000/api/v1/commands
until $(curl --silent --output /dev/null --head --fail ${volumio}); do
   echo "We wait till volumio is up and running"
   sleep 10s
done

echo "Volumio is running, so we restart the playlist"
curl localhost:3000/api/v1/commands/?cmd='playplaylist&name=PL_Classic21'
```
- Log generated during the reboot. As you can see, we get as response `{"time":1546344101724,"response":"playplaylist Success"}`
```
cat /var/log/cron.log
We wait till volumio is up and running
We wait till volumio is up and running
We wait till volumio is up and running
We wait till volumio is up and running
Volumio is running, so we restart the playlist
{"time":1546344101724,"response":"playplaylist Success"}
```
- Crontab definition
```
volumio@volumio:~$ cat /etc/crontab 
# /etc/crontab: system-wide crontab
# Unlike any other crontab you don't have to run the `crontab'
# command to install the new version when you edit this file
# and files in /etc/cron.d. These files also have username fields,
# that none of the other crontabs do.

MAILTO=""
SHELL=/bin/sh
PATH=/usr/local/sbin:/usr/local/bin:/sbin:/bin:/usr/sbin:/usr/bin
...

## Reboot and use playlist defined / default
@reboot volumio /home/volumio/start-playlist.sh >> /var/log/cron.log
```

cmoulliard, 2019-01-01:
I have added a sleep of `20s` before to call curl where we execute the `playplaylist` command and that works. 

See my instructions - https://github.com/ch007m/my-volumio-dev/blob/master/README.md#automate-the-restart-of-the-web-radio-using-a-playlist

cmoulliard, 2019-01-02:
Thanks for the heads up @xipmix . I will change the approach and use systemd. 

As mentioned in a different forum thread, the best approach should be to enhance the `auto-start` plugin to let a user to select within the configuration using a popup list, the playlists that they would like to select OR the list of web radios, ...

xipmix, 2019-01-02:
> As mentioned in a different forum thread

Could you link that here please for future reference?

cmoulliard, 2019-01-02:
See `Play a configured item` as defined here : https://volumio.org/forum/define-radio-per-default-started-t11374.html

cmoulliard, 2020-01-02:
What is the status about this issue ? Will it become part of a fix and released soon ? @xipmix

xipmix, 2020-01-03:
I have no idea, I'm afraid; @volumio takes these decisions. If we can figure out a patch and send a PR that should get accepted, eventually.

Let's define the feature you want more precisely. It sounds like you would like to see an 'auto-resume' control somewhere that allows you to tell volumio to start playing whatever is in the queue once it is fully started up. Correct? What's your thinking about which item under the 'cog' menu it should be added to? Possibly playback, could be restart, I'm not sure.

cmoulliard, 2020-01-03:
> It sounds like you would like to see an 'auto-resume' control somewhere that allows you to tell volumio to start playing whatever is in the queue once it is fully started up. Correct?

The use case is pretty simple. As a user, I would like that when volumio reboots, that it reconnects and plays to the last queue entry used before to stop. Such entry could by example my radio - ABCD).

cmoulliard, 2020-01-03:
> What's your thinking about which item under the 'cog' menu it should be added to? Possibly playback, could be restart, I'm not sure.

if it is not possible for the nodejs application to find the last entry to be played automatically at boot time, then we could imagine to have another entry under the playback menu, under the cog where the user could define what should the default web radio, ... to be played/started after a boot/reboot

xipmix, 2020-01-05:
This sounds pretty close to what the autostart plugin should be doing. I installed it and gave it a try;
added an album to playlist, rebooted partway through a track and it tried to play the interrupted track for me.
(There's some unrelated issue that prevented it actually working but not relevant here).

GOTO-GOSUB, 2020-12-16:
+1 here as well - all I want Volumio to do is play from an internet radio stream on startup. When I first set this up (on a Pi 3A+) and with the autostart plug-in installed it worked for 2 days and now every time the Pi restarts I have to go into the web GUI and hit the play button. I have ramped up the autostart delay to no avail. There are no networked or USB drives attached, no media servers. Just a couple of radio stations in my favorites. I don't know what changed on day 3 but even with resetting to defaults and starting again I cannot get this to work at all now. What I want the product do do sounds so simple - please, has anyone got this working ?

AndreiPuchko, 2022-01-15:
just never stop :)
```bash
#/bin/sh
volumio=localhost:3000/api/v1/commands
until $(curl --silent --output /dev/null --head --fail ${volumio}); do
   echo "We wait till volumio is up and running" >>aplay.txt
   sleep 5s
done

until false;
do
  if curl --silent localhost:3000/api/v1/getState | grep -q '"status":"play"'
  then
     echo "play" >>aplay.txt
     break
  else
     echo "starting..." >>aplay.txt
     curl localhost:3000/api/v1/commands/?cmd=play >>aplay.txt
     sleep 5s
  fi
done
echo "done" >>aplay.txt
```