volumio/Volumio2 #1696: [cli] Add commands to volumio cli (load playlist, load web radio)

arthurlutz, 2019-01-07:
It would be nice and useful (in particular for use with https://github.com/nanomesher/Nanomesher_NanoSound/ ) to be able do more with the volumio cli : 

* `volumio playlist load [1-9]`
* `volumio togglesound` (mute/unmute)
* `volumio radio load [1-9]`

xipmix, 2019-01-08:
Interesting. 'toggle' is implemented but not documented in the help. It just pauses/unpauses playback.
There is a 'volume' command that has the subcommands 'mute' and 'unmute', but no 'toggle'.

With the 'load' commands you are proposing, what do the numbers refer to? Do you want e.g.
```volumio playlist load 4``` to play the 4th track in the current volumio queue?

xipmix, 2019-01-08:
Curiouser and curiouser - 'volumio volume toggle' works as expected. Could you please test & confirm?

arthurlutz, 2019-01-13:
@xipmix thanks for the answer. I'll take a look at the code and see if I can contribute a patch to the help message of the cli to get `volumio volume toggle` displayed (and update Nanomesher_nanosound).

For the playlist load, I was thinking more of loading the first playlist or the second playlist etc. Maybe the names of the playlist can also be used, but keep in mind I am thinking of a use case with a remote control (which has 1 to 9 buttons).

xipmix, 2019-01-14:
>  I'll take a look at the code and see if I can contribute a patch to the help message of the cli 
I think that is done now, see #1699.

I see now why the numbers are a natural way to reference the playlists but I wonder what sets the enumeration and whether it is stable across reboots. If I understand, you're expecting to fetch the playlists displayed by viewing (Browse->Playlists) and get them in the order shown there?

arthurlutz, 2019-01-14:
@xipmix that sounds like a good summary/translation.

maxmib, 2019-05-22:
Is there any progress about 'volumio playlist load'?Recently I try to use an IR controller to control
volumio on my raspberry 3b.I use mpc command to load playlist ,like 'mpc load music '(music is the name of my playlist).Everything works fine but lircd stucks when  I use mpc commands and volumio commands together.   For example,the volumio playlist is empty,I use mpc  to load a playlist through  Ir controller,then if I use volumio toggle ,lircd stucks.I think it would be a good idea to enhance  volumio control function by adding commands similiar to mpc's .