volumio/Volumio2 #1710: Podcast plugin doesn't handle long podcast very well

PkAIR, 2019-01-27:
I'm using **Podcast** plugin version **0.1.0** with latest version of volumio (**2.526**) and currently experiencing issues with playing rather long episodes (~ 300mins). For instance - www.radiorecord.ru/rss.xml. It can be added successfully via the plugin configuration and even a particular episode can be started. THe issue is that after playing the episode for 1.5 mins sound simply disappears (but countdown updating).

xipmix, 2019-01-27:
Thanks for reporting. However this issue belongs with the [volumio/volumio-plugins ](https://github.com/volumio/volumio-plugins)sub-project. Can you please open an issue for this there, so the plugin maintainer will see it?

PkAIR, 2019-01-27:
Dag nam it. Sure thing, will do. Thanks!

closed 2019-01-27 by PkAIR