volumio/Volumio2 #1713: NFS mount and NAS disk hibernation/sleep

pecuchet, 2019-02-09:
Hello, Volumio is doing great here as DLNA renderer on RPI with my music mounted through NFS. However as long as the pi is running and although I don't listen to music, the nas will never go into hibernation mode. I suppose this is inherent to mounting through nfs, but it can maybe be circumvented through a kind of schedule/listening for activity? Or is there some other way?

In times of climatic challenges, this would be a great feature - to cut down power consumption... ;)

xipmix, 2019-02-12:
One approach is to use an automounter from volumio (which will auto-unmount after some period of inactivity) but this is a lot more complicated to configure and not so reliable in a home network.

An alternative would be for volumio itself to go to sleep after a period of inactivity:
 * in the Sleep options, add a new one - auto-shutdown after N minutes. This option should be off by default and if N==0.
 * a timer for this runs constantly, counting down from N
 * each new 'play' event restarts the timer
 * if the timer reaches 0, volumio shuts down

xipmix, 2019-02-12:
Perhaps you could try an experiment - use the existing volumio sleep function and observe whether your NAS is able to seel as well. Can you please try and report back? If that works then maybe the auto-sleep idea is worth pursuing.