volumio/Volumio2 #1716: Adding album to playlist adds entire Artist discography

livieris, 2019-02-13:
V 2.522.  Raspberry pi 3B
I also tried it with V 2.527.
When I try adding an album of an artist to new or existing playlist, it'll add the entire artists discography. The only work around is to add each individual song.
To reproduce, select an artist with more then one album, bring up the quick menu for an album, add to playlist. It doesn't matter if the playlist exists or if it's a new one. After adding, go to that playlist, you should see every album for that artist.

xipmix, 2019-02-14:
Thanks for reporting this, I will try to reproduce. Are you aware of a version where this did not occur?

xipmix, 2019-02-14:
Tested on 2.513 and it doesn't do that.
 * I had a small library on a NAS with two Van Morrison albums (and some other artists).
   * Selected one album, add to a new playlist. Works as expected, just adds the tracks from that album.
   * Deleted playlist, selected other album, added to a new and differently named playlist. Works as expected, just adds the tracks from that album.
   * Tried one album with an existing playlist. Works as expected, just adds the tracks from that album.

Can you please add [Regression] to the title of this issue?

volumio, 2019-02-14:
Might be due to this commit: 
https://github.com/volumio/Volumio2/commit/73603df2a9106c7b1eac6b3cc86c9b00610d37a9

@xipmix could you please check by reverting such commit?

xipmix, 2019-02-14:
I have 2.513 running and tried _applying_ that commit. I don't see the behaviour. I have to stop now.

When I get a little time I will try reverting it in 2.527, @livieris please feel free to try yourself and letting us know.

livieris, 2019-02-14:
I'll add some more info that I left out because I didn't think it would matter to this case.  My entire library is on a flash drive, ~17,000 songs. I am also using it with a waveshare 7in touchscreen.

xipmix, 2019-02-16:
Tried to reproduce again, with a fresh 2.526 image. Can't do it.
 * Set up my test music collection, let the scan complete.
 * Added one album to a new playlist that did not exist before I started. Playlist shows just the content of that album. 
 * Added a different album (same artist) to a different playlist. Playlist shows just the content of that album.
 * cleared all but one song from one of the new playlists above and tried adding an album again (same artist) to that existing playlist. Playlist got just the content of that album added.

Hm. The large collection is the most significant difference between our setups. Can you make a smaller drive with say 3 artists and a few albums each? That would rule the large collection out and make analysing logs more tractable. If you can reproduce with a smaller collection, can you send some logs?
 * reboot
 * add an album to a new playlist
 * send the logs https://volumio.github.io/docs/User_Manual/Sending_logs_for_troubleshooting.html

chsims1, 2019-02-16:
I think this depends how the album is added.  If I add an album via Music Library/NAS etc. then it adds the selected album only.  However, if I add an album via Artists, then all the artist's albums are added to the playlist.

livieris, 2019-02-18:
Sorry for the late response, was busy all weekend. I will try that tonight. Also didn't know there was more then one way to add an album to a playlist.

xipmix, 2019-02-19:
No worries, please let us know which method you used

livieris, 2019-02-19:
Before work this morning I was able to test using the Music Library/usb/artist and selected an album to add. When doing it this way, it adds only that album. When doing it from Artist/Album, it adds all of the albums. This is with my huge library still.

livieris, 2019-02-20:
I may have figured out the bug. Maybe. I noticed this when I went to add a few pink floyd albums to my playlist and saw every album was in it.  So at first it worked from Music Library/usb/artist.  It failed when I would add an artist with 2 more albums and try and add a single album from another artist. 
To reproduce: Add a single album to a playlist, add an artist with 2 or more albums to a playlist, then add a single album from another artist to the playlist.
 I tested again adding random selected albums, then I added all 3 doors down from the artist page. Then I went to select an album from A Perfect circle. Here is the log http://logs.volumio.org/volumio/wCQT6Nf.html.

xipmix, 2019-02-21:
Thanks for your work on this. I'll try to reproduce as you describe.
When you add albums to a playlist, do you always go via Browse->MusicLibrary->USB->Artist Name->Album, or Browse->Artists->Artist Name->Album? 
From what you've written above it seems like the former but I just want to clarify in case I have difficulty reproducing.

livieris, 2019-02-21:
At first I always did Browse/Artist/Album. Which always failed no matter what. Just tested this way again by adding a single album and it adds every album.
Then I tried it from Browse/Music Library/Artist/Album because at least this way I could add a single album until I found the bug above. So when trying to reproduce, use Browse/Music Library/usb/artist.

xipmix, 2019-02-26:
Ok got it to reproduce. Existing playlist with 1 track in it. Browse->Artists-><artist name>. From the menu to the right of the first album, choose 'add to playlist'. It adds all tracks from both albums in my test collection.

The entirety of what was logged in /var/log/volumio.log when I pressed 'add to playlist' is below:
```
2019-02-26T12:10:08.595Z - info: Listing playlists
2019-02-26T12:10:10.565Z - info: Pushing Favourites {"uri":"artists://Van%20Morrison/Inarticulate%20Speech%20Of%20The%20Heart","favourite":false}
2019-02-26T12:10:10.571Z - info: Listing playlists
2019-02-26T12:10:11.351Z - error: error
2019-02-26T12:10:11.504Z - error: error
2019-02-26T12:10:12.567Z - error: error
2019-02-26T12:10:12.737Z - error: error
2019-02-26T12:10:19.280Z - info: CoreCommandRouter::executeOnPlugin: mpd , handleBrowseUri
2019-02-26T12:10:19.282Z - info: CURURI: playlists
2019-02-26T12:10:19.283Z - info: Listing playlists
```

xipmix, 2019-02-26:
On a quick look I can't see 73603df in the code, which is really odd. Here are the lines around where it should be.
```
$ cd /volumio; less -N app/plugins/music_service/mpd/index.js
...
   2054     else if(uri.startsWith('albums://')) {
   2055         //exploding search
   2056         var splitted = uri.split('/');
   2057         var artistName = decodeURIComponent(splitted[2]);
   2058         var albumName = decodeURIComponent(splitted[3]);
   2059         var cmd = libMpd.cmd;
   2060         // Escape any " within the strings used to construct the 'find' cmd
   2061         var safeArtistName = artistName.replace(/"/g,'\\"');
   2062         var safeAlbumName = albumName.replace(/"/g,'\\"');
   2063 
   2064                 if (compilation.indexOf(artistName)>-1) {  //artist is in Various Artists array
   2065                         var GetAlbum = "find album \""+safeAlbumName+"\"" + " albumartist \"" +safeArtistName+"\"";
   2066                 }
   2067                 else {
   2068                         var GetAlbum = "find album \""+safeAlbumName+"\"" + " artist \"" +safeArtistName+"\"";
   2069                 }
   2070 
   2071                 self.clientMpd.sendCommand(cmd(GetAlbum, []), function (err, msg) {
   2072             var list = [];
   2073             if (msg) {
   2074                 var path;
   2075                 var name;
```
Here's the place that git says we are up to
```
$ git log -1
commit 78de34390ebc5df0b086e244ca969a4e0ddc40d1
Author: Volumio <info@volumio.org>
Date:   Fri Jan 11 15:21:46 2019 +0000

    handle non primo devices

$ git log -1 --  app/plugins/music_service/mpd/index.js
commit 70bd89892235d63be5c6faf22573c86838181362
Author: Volumio <info@volumio.org>
Date:   Sun Sep 16 01:46:13 2018 +0000

    Ability to delete folder in internal memory
```
No idea what is going on there.

But that aside, I don't think 73603df is the problem as that section of the code deals with uris starting with ```albums://```. Clearly here we are trying to add an album but the uri has been prefixed with ```artists://```. Presumably this is a bug in the Volumio2-UI code, or how the menu I am selecting from gets filled out.

xipmix, 2019-02-26:
The missing 73603df is because it went in after 2.526 was released, doh. 
The issue reported here might also be fixable by checking whether an ```artists://``` uri has an album name appended, after the artist and in that case limiting what is added to the playlist to just that album.

livieris, 2019-02-26:
So I may have found a workaround. Which may also help with this bug as I imagine the code for this process is similar. So I tried just adding to queue with what I did before. Add an album, add all artist, add single album. Seemed to work. And from there you can create a playlist out of that queue. But more important doesn't seem to fail. I will try more tests when I get the chance tonight and update.

xipmix, 2019-02-27:
As far as I can see the problem lies here in app/playlistManager.js:
```
    392 PlaylistManager.prototype.commonAddToPlaylist = function (folder, name, service, uri, title) {
...
    436                 else if(uri.startsWith("artists://"))
    437                 {
    438                     mpdPlugin.listArtist(uri,2,'')
    439                     .then(function(entries){
    440                         listingDefer.resolve(entries.navigation.lists[1].items);
    441                     }).fail(function(){
    442                         listingDefer.reject(new Error());
    443                     })
    444                 }
```

This needs to be doing the same check as in the mpd plugin where it calls ```listAlbumSongs``` if the uri contains an album name after the artist name.

I tried this patch to the file in 2.526 (watch for line number differences) but it does not work
```
@@ -426,12 +427,25 @@
                 }
                 else if(uri.startsWith("artists://"))
                 {
+self.logger.info("commonAddToPlaylist: "+uri);
+var s = uri.split('/');
+self.logger.info("got s:"+s);
+                    if (s.length==3) { // no album name
                     mpdPlugin.listArtist(uri,2,'')
                     .then(function(entries){
                         listingDefer.resolve(entries.navigation.lists[1].items);
                     }).fail(function(){
                         listingDefer.reject(new Error());
                     })
+                    } else {
+self.logger.info("calling listAlbumSongs:");
+                    mpdPlugin.listAlbumSongs(uri,2,'artists://' + s[2] + '/' + s[3])
+                    .then(function(entries){
+                        listingDefer.resolve(entries.navigation.lists[1].items);
+                    }).fail(function(){
+                        listingDefer.reject(new Error());
+                    })
+                    }
                 }
                 else if(uri.startsWith("albums://"))
                 {
```
I tested with Browse->Artist->(Artist Name)->click on menu next to album, add to (new) playlist.
I end up with a new playlist but it is empty.

What I get in the log is this, which looks ok but as I say the playlist ends up empty.
```
2019-02-27T10:16:10.977Z - info: commonAddToPlaylist: artists://Van%20Morrison/Inarticulate%20Speech%20Of%20The%20Heart
2019-02-27T10:16:10.979Z - info: got s:artists:,,Van%20Morrison,Inarticulate%20Speech%20Of%20The%20Heart
2019-02-27T10:16:10.979Z - info: calling listAlbumSongs:
```

xipmix, 2019-02-27:
Digging a bit futher
 * the uri makes it into listAlbumSongs() unscathed
 * the ```findstring``` it feeds to mpd is correct:
 ```find album "Inarticulate Speech Of The Heart" artist "Van Morrison"```
  * The artist and album fields are returned correctly, the correct number of tracks are found
  * The response object constructed by the function looks fine, only one album's worth of tracks is included. There are two fields named ```uri:``` (same strng  in both), could that be an issue?

So this is probably some javascript strangeness I don't understand.

xipmix, 2019-02-27:
I spotted it finally, I needed to refer to ```lists[0]``` not ```lists[1]``` when handling the result from ```listAlbumSongs()```.

extrapilot, 2019-07-22:
Hey guys

Sorry, no idea if this was supposed to be corrected for the current release, or if it remains on the known bug list. But at least for a NAS-based source, this problem still exists, where adding an album via Artists->Artist entry->Album, where selecting the Add To Playlist there from the 3 dot menu pulls in the entire catalog of that artist under the scope of whatever Playlist name you define. If instead you search by Album directly- then the same process yields only that album’s tracks in the playlist. Perhaps this behavior is different for a locally-mounted catalog. This is on an RPI3 running whatever variant is defined as ‘current’ as of 7.22.19 (I don’t see the version number in System or Credits).

xipmix, 2019-08-26:
@extrapilot thanks for commenting. Are you able to test the fix in  PR #1728 ? If you can comment there that it Works For You that might help get the fix merged.

PR #1728 (2019-10-01): Correct handling of artists:// uri with an album appended
Fixes: #1716

Tested on rpi2 and release 2.526.
closed 2019-10-01 by volumio