volumio/Volumio2 #1726: security - REST API

koenbuyens, 2019-02-23:
Currently the REST API endpoint for backup returns the passwords of my shared windows drive, spotify, and WiFi without authentication. This enables someone who is on the same network as volumio to obtain those passwords. For instance, if I give one of my friends my WiFi password when visiting, they could also obtain my Spotify password or the password of my shared drive. This is not ideal.

Steps to reproduce:
1. Go to http://<volumio.local>/api/v1/backup/config/ as an unauthenticated user. (replace <volumio.local> with your instance.)
2. Observe that the response contains the passwords mentioned.

What I would like to see:
An Authorization token in the Authorization header. If an incorrect token is present, the API rejects the request. If the correct token is present, it accepts it.
Ideally, the admin user can enable this in the settings page (and the settings page requires an admin password to access).

volumio, 2019-02-26:
You're right. Thanks for the report, we'll add this security layer.

volumio, 2019-02-27:
For the moment, we disabled those endpoints (they were made but never used for anything useful). 
We will restructure and add an authentication method

closed 2019-02-27 by volumio