volumio/Volumio2 #1730: Volumio will not play Spotify playlist

dai-kan, 2019-03-01:
With Volumio ver 2.555 installed, Spotify ver 1.35, Modified index.js file with Spotify refresh token per directions in member2398.html, afterwards Spotify playlist is visible in Browse-Spotify-My Playlist-Playlist1. Playlist1 was created in Spotify. The problem now is when I click on the item to play, nothing happens. Likewise the using play in right hand drop down box does not work either. Please help.

wibenjamin, 2019-03-03:
I can confirm this problem too. I tried the latest Volumio Release and also V2.411 and V2.389 also with the modiefied index.js file.  Access and authentification via the refresh token seems to work, playback of single tracks from spotify search is working, also tracks and playlists from the "Whats new" tab also appear to work. All the other playlists are showing fine but they can not be accessed or played or added to the queue.

daveschafer, 2019-03-13:
I can also confirm, that this issue still persists in Version 2.565

Csierra099, 2019-03-15:
Indeed the problem still persits, is there any solution out there or maybe an alternative? I'm the IT guy from an hotel and we have 3 Pi's running Volumio but since this problem started going on we've been having a lot of issues trying to put music.

Baudouin0, 2019-04-07:
Any plan to fix this?

dai-kan, 2019-04-07:
I posted this as an issue.  Perhaps someone from Volumio can pick it up and find a solution.

wibenjamin, 2019-04-08:
The developer of the plugin fixed the issue and it is working fine for me.

Check this thread and just follow his instructions:
[https://forum.volumio.org/spotify-playlists-not-working-anymore-t11918.html#p61273](url)

xipmix, 2019-04-09:
If this is really fixed, can you please close this issue?

Baudouin0, 2019-04-09:
Did clone the latest git branch and installed manually into volumio. Still cannot get any sound from genres and modes, the only thing which plays are the new releases.

closed 2019-04-09 by dai-kan

b-d-k, 2019-05-29:
This has unfortunately not been fixed. There is a somewhat 'hacky' fix in this fork: https://github.com/skikirkwood/volumio-plugins, but the changes are not in this repo, nor could I find a PR for it.

koesper, 2019-12-30:
are there any updates on this? still doesnt work with the 'official' versions, right?

volumio, 2019-12-30:
Yes we fixed it. Update volumio to latest version and update spotify to latest version. Then authorise it