volumio/Volumio2 #1731: Compatibility with ReSpeaker 2-Mics Pi HAT

dubaleeiro, 2019-03-10:
Hi,
I have successfully installed Volumio in a Raspberry Pi Zero W with the above ReSpeaker 2-Mics Pi HAT but it seems it is not supported.
I cant make any audio to be played throught the Audio JAC or JST connectors for speakers.
Could you please help?
Thanks.

Ultchad, 2019-03-30:
I have the same problem

I have installed the header via `volumio kernelsource`
install kernel modul and overwrite with https://github.com/respeaker/seeed-voicecard
(No apt-get upgrade or dist-upgrade during the install, I know i already break volumio)

and i see i2c with i2cdetect but no card with `aplay -l` or `arecord -l`

After i add to dacs.json file, and aply:
```json
{"id":"seeed-2mic-voicecard","name":"seeed-2mic-voicecard","overlay":"seeed-2mic-voicecard","alsanum":"1","mixer":"","modules":"snd-soc-seeed-voicecard","script":"","i2c_address":"1a","needsreboot":"yes"}
```

and i lose i2c

Now i don't know what i can test ( on raspbian it's work great )

dubaleeiro, 2019-04-04:
Sorry to say that... but isnt there an easier way?