volumio/Volumio2 #1747: Mixer Type undefined Issue

thegunslingers, 2019-04-07:
Setup: 
Volumio on a PiZero
USB DAC 
Hardware mixer type

Everything works as it should, but occasionally with the USB DAC turned off the mixer type will be changed to undefined and requires an edit to the playback options to fix. I have tested restarting Volumio (with the DAC off) and verified that is one case where the mixer type changes from hardware to undefined.

xipmix, 2019-04-09:
So ... you have a hardware mixer, you configure volumio to use it and then you(?) turn it off. If not by you, how does it get turned off - a powersave feature?

If you turn the DAC on again and bring volumio up, what happens? 
It sounds like volumio does not notice, and does not fix the Mixer Type.

What do you think should happen here - should volumio automatically change to the software mixer? Or should Mixer Type change to some other defined value, e.g. 'not_detected' ?

thegunslingers, 2019-05-02:
apologize for the delayed response.
If I get Mixer Type to undefined, then turn on the dac then turn on volumio, mixer state stays as undefined, like you were thinking.
Ideally in my opinion it would stay 'hardware mixer' with 'mixer control name' saved as it was set until manually changed to be something else, even if the mixer is not detected or not plugged in.  I think it would be a little more robust with this setup.  And bad boot up order, or turning things off would not force a settings adjustment.

Thank you!

thegunslingers, 2019-05-16:
I have been working on a fix for this issue, and I have some working code, but I do have one JS question, before I try to submit the fix.
I want to check the config file to ensure mixer is not undefined or empty, I looked at other sections of the code and followed what I saw. The if statement appears to execute everytime though even when my alsa config.json has
`  "mixer": {
    "type": "string",
    "value": ""
  },`
`ControllerAlsa.prototype.setDefaultMixer  = function (device) {
	var self = this;
        var defaultmixer = self.config.get('mixer', '')
        if(defaultmixer.value != undefined && defaultmixer!=''){ 
                    ...js fix here...
        }`

thegunslingers, 2019-05-18:
Submitted pull request with fix

xipmix, 2020-01-04:
Hi, is this resolved for you - if so could you close it please?

thegunslingers, 2020-01-04:
thanks the hot plug changes now fix this issue

closed 2020-01-04 by thegunslingers