volumio/Volumio2 #1751: Rest API playlist play fails

eloquentvogon, 2019-04-13:
Using the API I am calling a playlist that comes back from the listplaylist call, however the command below does not load the playlist 

`curl -i '192.168.0.150/api/v1/?cmd=playplaylist&N=dog'
HTTP/1.1 200 OK
X-Powered-By: Express
Access-Control-Allow-Origin: *
Access-Control-Allow-Methods: GET,PUT,POST,DELETE,OPTIONS
Access-Control-Allow-Headers: Content-Type, Authorization, Content-Length, X-Requested-With
Content-Type: application/json; charset=utf-8
Content-Length: 36
ETag: W/"24-goTtfPJHCnKHD/2uO3Ad68RaujQ"
Vary: Accept-Encoding
Date: Sat, 13 Apr 2019 23:45:57 GMT
Connection: keep-alive

{"message":"Welcome to Volumio API"}`

Nothing related in the logs

petershaw23, 2019-05-13:
you need additional paranthesis!
like so:
`'curl 192.168.0.178/api/v1/commands/?cmd="playplaylist&name=playlist6"'`
if it is inside another argument or like this:
`curl 192.168.0.178/api/v1/commands/?cmd='playplaylist&name=playlist6'`
if it's a single line command

eloquentvogon, 2019-05-26:
Thank you, I will give this a try